AltBeacon/android-beacon-library #68: model distance calculations

Fiodora, 2015-01-03:
Hi,
i made some RSSI measurements for my device. I wanted to ask is there a way to change the model-distance-calculations.json in the .aar file?
```
My measurements

Meters  RSSI
0.25    -48
0.5     -50
1       -64
2       -58
3       -61
4       -66
5       -68
6       -71
7       -74
8       -76
9       -78
10      -79
12      -83
14      -85
16      -89
18      -90
20      -90
25      -91
30      -92
40      -93

Android device:
version: 4.4.2
build_number: KOT49H.T535XXU1ANK1
model: Galaxy Tab 4
model_number: SM-T535
manufacturer: SAMSUNG

Beacon Info:
Beacon inside B0001-A
Advertising Interval: 400ms
Transmit Power: 0dBm
```

```
Ratio;Distance
0.75;0.25
0.78125;0.5
1;1
0.90625;2
0.953125;3
1.03125;4
1.0625;5
1.109375;6
1.15625;7
1.1875;8
1.21875;9
1.234375;10
1.296875;12
1.328125;14
1.390625;16
1.40625;18
1.40625;20
1.421875;25
1.4375;30
1.453125;40
```
My Calculation
```
[1] "Imported Data"
      Ratio Distance
1  0.750000     0.25
2  0.781250     0.50
3  1.000000     1.00
4  0.906250     2.00
5  0.953125     3.00
6  1.031250     4.00
7  1.062500     5.00
8  1.109375     6.00
9  1.156250     7.00
10 1.187500     8.00
11 1.218750     9.00
12 1.234375    10.00
13 1.296875    12.00
14 1.328125    14.00
15 1.390625    16.00
16 1.406250    18.00
17 1.406250    20.00
18 1.421875    25.00
19 1.437500    30.00
20 1.453125    40.00

4035.433 :  0 1 2
4022.721 :  0.4978636 0.5105707 3.1976115
3883.051 :  0.7761659 0.2603231 5.4045057
3282.947 :  0.9189422 0.2073381 8.1175731
2304.326 :  1.1298147 0.2254358 9.8166507
1156.337 :   1.564938  0.253328 11.064331
320.7118 :   2.3366290  0.2643383 12.1453272
159.3244 :   3.3149051  0.2192556 13.1973542
158.9603 :   3.4108690  0.1950424 13.4966767
158.6838 :   3.485017  0.176806 13.760706
158.4035 :   3.5447094  0.1625738 13.9919690
158.2493 :   3.6432924  0.1398329 14.3959770
157.7733 :   3.7095553  0.1259602 14.6986374
157.6137 :   3.7577786  0.1163706 14.9223985
157.5388 :   3.792645  0.109707 15.088256
157.503 :   3.8180570  0.1049938 15.2110883
157.4851 :   3.8366407  0.1016241 15.3019644
157.4759 :   3.8502605  0.0991957 15.3691271
157.471 :   3.86025574  0.09743584 15.41871527
157.4684 :   3.86759748  0.09615518 15.45529964
157.467 :   3.87299259  0.09522054 15.48227055
157.4663 :   3.87695936  0.09453687 15.50214673
157.4659 :   3.87987600  0.09403606 15.51678641
157.4657 :   3.88202119  0.09366875 15.52756713
157.4656 :   3.88359896  0.09339913 15.53550399
157.4655 :   3.88475956  0.09320111 15.54134612
157.4655 :   3.88561328  0.09305561 15.54564546
157.4654 :   3.88624118  0.09294868 15.54880887
157.4654 :   3.88670324  0.09287005 15.55113725
157.4654 :   3.88704329  0.09281221 15.55285112
157.4654 :   3.8872933  0.0927697 15.5541112
157.4654 :   3.88747725  0.09273842 15.55503862
157.4654 :   3.88761248  0.09271542 15.55572099
157.4654 :   3.88771221  0.09269848 15.55622329
157.4654 :   3.88778536  0.09268605 15.55659207
157.4654 :   3.88783925  0.09267689 15.55686395
157.4654 :   3.88787902  0.09267013 15.55706462
157.4654 :   3.88790809  0.09266519 15.55721131

[1] "Power Curve Coefficients"
  intercept  multiplier       power
 3.88790809  0.09266519 15.55721131
```
Could i use them like this?
```
 {
"coefficient1": 3.88790809,
"coefficient2": 0.09266519,
"coefficient3": 15.55721131,
"version":"4.4.2",
"build_number":"KOT49H",
"model":"SM-T535",
"manufacturer":"SAMSUNG"
},
```

davidgyoung, 2015-01-03:
These measurements and coefficients look reasonable.  I have put these into model-distance-calculations.json in a snapshot build available here:

https://bintray.com/altbeacon/android/android-beacon-library/2.1-beta1-16-gf1af660/view/files/org/altbeacon/android-beacon-library/2.1-beta1-16-gf1af660

@Fiodora, please try this version with the device and report back if you get good distance estimates with this in place.  If so, I can publish a json update on a server so other folks will have access.

Fiodora, 2015-01-06:
I tried using your snapshot build, but somehow im not able to use the calculations for my Model.
I printed the used Model on my Screen, i even tried to force to use my Model.
```
 AndroidModel model = new AndroidModel("4.4.2","KOT49H","SM-T535", "SAMSUNG");
        ModelSpecificDistanceCalculator mod = new ModelSpecificDistanceCalculator(getApplicationContext(), "a", model); // Im not quite sure what the url should be
        logToDisplay("" + mod.getModel() + "req:" + mod.getRequestedModel());
```

davidgyoung, 2015-01-16:
Sorry for the delay in responding.  

I suspect the reason your model wasn't being selected automatically is because I had entered the build number "KOT49H" instead of "KOT49H.T535XXU1ANK1" and the manufacturer as "SAMSUNG" not "Samsung".  I have changed the manufacturer check to not be case sensitive, and updated the model number.   Here's a new shapshot build to try:

https://bintray.com/artifact/download/altbeacon/android/org/altbeacon/android-beacon-library/2.1-beta1-18-gec95398/android-beacon-library-2.1-beta1-18-gec95398.aar

If you find this snapshot gives better distance estimates on your device than the default formula with the regular version of the library, please let me know.

I worry that the 1, 2 and 3 meter measurements you took may be a problem.   Both the 2 and 3 meter measurements represent a stronger signal than the 1 meter measurement.  I have no doubt you actually measured these values, but they are probably overly polluted by noise or reflections.  While this can sometimes happen (especially with the measurements at the farther distances) you are unlikely to get a working model if the signals you measure for under 5 meters don't represent progressively weaker signals as you move away.

MGasztold, 2015-04-09:
Where can I find complete model-distance-calculations.json file with all the models?

davidgyoung, 2015-04-09:
@MGasztold, the models are all present here:  https://github.com/AltBeacon/android-beacon-library/blob/master/src/main/resources/model-distance-calculations.json

Obviously, the list is very sparse.  This is something that has been in sore need of attention for a long time.  I am very interested in a volunteer with decent math skills to help update the equations in our list.

davidgyoung, 2018-03-06:
Closing due to inactivity

closed 2018-03-06 by davidgyoung