AltBeacon/android-beacon-library #71: Class load warnings on Android 4.4 and lower

markvandertol, 2015-01-04:
I compiled the `master` branch and tried it on my Android 4.4 phone, but it shows a lot of warnings on LogCat. It still works, but useless warnings are of course undesired. 

```
01-04 21:03:34.111 E/dalvikvm﹕ Could not find class 'android.bluetooth.le.ScanCallback', referenced from method org.altbeacon.beacon.service.CycledLeScanner.finishScanCycle
01-04 21:03:34.111 W/dalvikvm﹕ VFY: unable to resolve check-cast 32 (Landroid/bluetooth/le/ScanCallback;) in Lorg/altbeacon/beacon/service/CycledLeScanner;
01-04 21:03:34.121 W/dalvikvm﹕ Unable to resolve superclass of Lorg/altbeacon/beacon/service/CycledLeScanner$4; (32)
01-04 21:03:34.121 W/dalvikvm﹕ Link of class 'Lorg/altbeacon/beacon/service/CycledLeScanner$4;' failed
01-04 21:03:34.121 E/dalvikvm﹕ Could not find class 'org.altbeacon.beacon.service.CycledLeScanner$4', referenced from method org.altbeacon.beacon.service.CycledLeScanner.getNewLeScanCallback
01-04 21:03:34.121 W/dalvikvm﹕ VFY: unable to resolve new-instance 452 (Lorg/altbeacon/beacon/service/CycledLeScanner$4;) in Lorg/altbeacon/beacon/service/CycledLeScanner;
01-04 21:03:34.121 W/dalvikvm﹕ VFY: unable to resolve virtual method 60: Landroid/bluetooth/BluetoothAdapter;.getBluetoothLeScanner ()Landroid/bluetooth/le/BluetoothLeScanner;
01-04 21:03:34.121 E/dalvikvm﹕ Could not find class 'android.bluetooth.le.ScanSettings$Builder', referenced from method org.altbeacon.beacon.service.CycledLeScanner.scanLeDevice
01-04 21:03:34.121 W/dalvikvm﹕ VFY: unable to resolve new-instance 35 (Landroid/bluetooth/le/ScanSettings$Builder;) in Lorg/altbeacon/beacon/service/CycledLeScanner;
01-04 21:03:34.121 E/dalvikvm﹕ Could not find class 'android.bluetooth.le.ScanSettings$Builder', referenced from method org.altbeacon.beacon.service.CycledLeScanner.scanLeDevice
01-04 21:03:34.121 W/dalvikvm﹕ VFY: unable to resolve new-instance 35 (Landroid/bluetooth/le/ScanSettings$Builder;) in Lorg/altbeacon/beacon/service/CycledLeScanner;
01-04 21:03:34.121 E/dalvikvm﹕ Could not find class 'android.bluetooth.le.ScanCallback', referenced from method org.altbeacon.beacon.service.CycledLeScanner.stop
01-04 21:03:34.131 W/dalvikvm﹕ VFY: unable to resolve check-cast 32 (Landroid/bluetooth/le/ScanCallback;) in Lorg/altbeacon/beacon/service/CycledLeScanner;
01-04 21:03:34.131 W/dalvikvm﹕ Unable to resolve superclass of Lorg/altbeacon/beacon/service/CycledLeScanner$4; (32)
01-04 21:03:34.131 W/dalvikvm﹕ Link of class 'Lorg/altbeacon/beacon/service/CycledLeScanner$4;' failed
```

The guards with `if` statements won't work to prevent these warnings. When you instantiate a class, all classes it directly depends on should also exist.

A workaround by putting the two variants in two different classes is explained in this Stackoverflow post: http://stackoverflow.com/a/11593151/1364738. I think it will also help for readability of the code.

If you agree I can split the L implementation and the "old" implementation in two classes and send in a pull request.

davidgyoung, 2015-01-05:
Thanks for this feedback @markvandertol, this is very useful!  I think your analysis is right on.  Yes, if you would like to send a pull request for this it would be great.

davidgyoung, 2015-01-13:
A fix for this has been merged into the 2.1-beta2 release.

closed 2015-01-13 by davidgyoung