AltBeacon/android-beacon-library #73: Build failed : Could not find property 'bintrayUsername'

pdavid0, 2015-01-08:
C:\Users\...\AndroidStudioProjects\android-beacon-library>gradlew.bat release --info

    Starting process 'command 'git''. Working directory: C:\Users\...\AndroidStudioProjects\android-beacon-library Command: git describe --tags
    Can't get version from git: org.gradle.process.internal.ExecException: A problem occurred starting process 'command 'git''
    Package resolve took 233 ms.                                                    
                            
    FAILURE: Build failed with an exception.
                            
    * Where:                    
    Script 'C:\Users\...\AndroidStudioProjects\android-beacon-library\gradle\bintray.gradle' line: 8
                            
    * What went wrong:          
    A problem occurred evaluating script.
    > Could not find property 'bintrayUsername' on com.jfrog.bintray.gradle.BintrayExtension_Decorated@4d5cef4c.

pieces029, 2015-01-08:
This was a bug on my part. I'll have a PR open in a second.

davidgyoung, 2015-01-08:
I put in placeholders in 0f2403462b567c2d4c6ccf07048c89ea09187e21 so this won't fail.  @pieces029, if you have a better solution let me know.

closed 2015-01-11 by davidgyoung