AltBeacon/android-beacon-library #78: Devices in Play Store must have BT but BT LE isn't required

markvandertol, 2015-01-11:
Because manifest specifies that it requires the permissions `android.permission.BLUETOOTH` and `android.permission.BLUETOOTH_ADMIN`, the Play Store implicitly requires that the device must support Bluetooth (nowadays almost all devices do, but still). However it doesn't implicitly require Bluetooth LE. As this library requires both to function this isn't ideal.

I see three possible solutions:

1. Explicitly require device support for Bluetooth LE. When requiring Bluetooth LE the app can no longer be installed on devices that don't support it. When beacons isn't part of your core app functionality then this isn't desired.
2. Explicitly don't require device support for Bluetooth. When not requiring Bluetooth it can be installed on all devices. When beacons are part of your core app functionality, this isn't desired.
3. Don't make changes to the required device capabilities. When these required capabilities are explicitly (not) required then you can no longer specify them in your app's manifest, as it would cause manifest merge errors. When implementing this library in your own app you can then specify whether you require Bluetooth and Bluetooth LE. This could be added to the quick start guide how to do this.

More info about implied permissions: http://developer.android.com/guide/topics/manifest/uses-feature-element.html#permissions

Do you also see this as an issue? I think option 3 is the best option as it is the most flexible.

davidgyoung, 2015-01-12:
My understanding is that you must put both of these permissions in the manifest in order for Android to not throw an exception when attempting to start a Bluetooth LE scan.  I am also not aware of any manifest permission specific to BLE (vs. Classic bluetooth.)

Do you think this is incorrect?

markvandertol, 2015-01-12:
I'm talking here about required device features. That is something different than permissions. Every device has a list of features it supports. You can specify those using `uses-feature` elements in your manifest.

When some permissions are defined than required device features are inferred, such is the case with the Bluetooth permission. The required feature list of an apk is only used by stores such as the Play Store to ensure the device has the required features to run the app. It isn't validated runtime whether the device has the features.

davidgyoung, 2015-01-13:
OK, sorry if I am a little slow to understand.  So what you are proposing as part of option 3 is to instruct users to put the following in their manifest if they don't want the PlayStore to filter out their apps from devices without bluetooth:

<uses-feature android:name="android.hardware.bluetooth" android:required="false" />

Sure, I think this is a good idea.  Since as you say this is very rare nowadays, perhaps this doesn't so much belong in the quick start guide (add too much info on edge cases and it isn't so quick!) but at least should be documented somewhere.

markvandertol, 2015-01-13:
I shall write a short page on the device features in the manifest related to Bluetooth. It is of course also advantages when you DO want to filter out devices when your app doesn't make sense without beacons.

davidgyoung, 2015-01-13:
OK, thanks.

davidgyoung, 2015-01-29:
The documentation for this has been merged into the project site.

closed 2015-01-29 by davidgyoung