AltBeacon/android-beacon-library #81: Release build defaulting to snapshot with wrong file name

davidgyoung, 2015-01-13:
after doing a `./gradlew release` with the tag 2.1-beta2, it looks like the version string is incorrectly getting suffixed with SNAPSHOT because I see:

    Building with version=2.1-beta2-SNAPSHOT

And then the files that get generated don't have the version at all in the suffix.  The contents of build/outputs/aar shows:

    android-beacon-library-release.aar	
    android-beacon-library-release.tar.gz

@pieces029, any thoughts here?

pieces029, 2015-01-13:
The snapshot gets appended if you don't pass gradle the -Prelease flag. I
can see why this would be confusing for a task and a flag to have the same
name.

pieces029, 2015-01-13:
We can change this up, it's just the way I do it since it's harder to accidentally publish something or check in the wrong version number this way. Thoughts?

davidgyoung, 2015-01-13:
Understood about the extra protection of the -Prelease flag.  That's fine by me.

Any thoughts on why the filenames don't become 

    android-beacon-library-2.1-beta2.aar  
    android-beacon-library-2.1-beta2.tar.gz

when I do:  `./gradlew release -Prelease` ?  Instead they are:

    android-beacon-library-release.aar	
    android-beacon-library-release.tar.gz

pieces029, 2015-01-13:
I think that's just how the assembleRelease task creates the aar's. I'll take a look and see if I can find anything about the gz files.

pieces029, 2015-01-13:
The tar file comes from bundle-eclipse script, basing it off of the name of the release aar. Did it not always use to be like this? I think the only compilation tasks I added where to create jar files of sources and javadocs.

davidgyoung, 2015-01-16:
Looking at this a bit more... yeah, this is not the way it used to be.  With the current setup, when you do a release build it always creates a local aar filename without a version number like this: /build/output/aar/android-beacon-library-release.aar .

When you upload it to bintray, it gets renamed to the version number, e.g. android-beacon-library-2.1-beta2.aar 

Since it gets renamed in the upload, that's all for the aar.  Except that the .tar.gz file (which is used for Eclipse and converted from the aar using the bundle-eclipse script) also ends up without a version number in the filename.  This is a problem because this file cannot be distributed on bintray and has to be manually uploaded to a release on Github so people can download it the old fashioned way.  So it has to be manually renamed to android-beacon-library-2.1-beta2.tar.gz first.

The way this worked up through fb5c11e20116302e510c457029a33a36e7be2720 was that a release build would produce a local filename with the version in it, e.g. android-beacon-library-2.1-beta1-4-gfb5c11e.aar, and the script that generates the tar.gz file would also get that same filename with a different extension android-beacon-library-2.1-beta1-4-gfb5c11e.tar.gz.  It would therefore be ready to upload to Github as a release.

Ideally, this process would still produce local build artifacts with those filenames for uploading to Github.  I'm not sure what caused this change, but one option might be to simply rename the files in an extra step at the end, or copy them to another build folder with those desired filenames.

pieces029, 2015-01-16:
That shouldn't be too hard to make a copy task with gradle to do this. I'll try and work one out over the weekend.

davidgyoung, 2015-01-18:
Fixed in #85

closed 2015-01-18 by davidgyoung