AltBeacon/android-beacon-library #83: Beacon distance is wrong

padcoe, 2015-01-15:
Hey buddy! How are you?

I've implemented this library but sometimes the distance calculated is wrong. Right now, my device (Galaxy S4) is 5.30 meters away from the ibeacon (a device from Kontact) and the method Beacon.getDistance() is ranging from 1.7469709545571706 to 2.4134186193894838.

Those values are in meters? Should i do any conversion to obtain the real distance(meters)?

Is possible to obtain a better distance's precision? I'm asking because i need show if the beacon is immediate, near or close (immediate <= 0.5, near > 0.5 && <= 3.0, far > 3.0).

Thank in advance.

davidgyoung, 2015-01-15:
The library must do a different calculation for each Android device model to get accurate distance estimates, and nobody has yet done the measurements for the Galaxy S4.  You can read more about how to do this here: 

http://altbeacon.github.io/android-beacon-library/distance-calculations.html

Would you be willing to take some measurements with your Galaxy S4 at the various distances and crunch the numbers?  If so, you can see how to do so here:  https://github.com/AltBeacon/android-beacon-library/issues/47

padcoe, 2015-01-16:
Good idea but i'll do for Motorola G...Samsung S4 is not mine, thank you.

davidgyoung, 2015-01-16:
@padcoe Thanks for the offer to do Moto G!  We actually do have measurements for that model, it's just that nobody has taken the time to crunch the numbers yet to get the coefficients.  If you'd be game for doing this, here they hare:

Meters	Rssi
0.25	-54
0.5	-62
1	-66
2	-75
3	-83
4	-83
5	-90
6	-89
7	-85
8	-81
9	-87
10	-84
12	-91
14	-94
16	-93
18	-95
20	-97
25	-95
30	-95
40	-97

jobinjamez, 2015-03-05:
@davidgyoung Could you let me know how to get the coefficients from these numbers. I could try doing it.

padcoe, 2015-03-05:
@jobinjamez i'm looking for it too.

padcoe, 2015-03-05:
MotoG

Meters 	Rssi
0.25	        -54
0.5 	         -62
1 		-66
2 		-75
3 		-83
4 		-83
5 		-90
6 		-89
7 		-85
8 		-81
9 		-87
10 		-84
12 		-91
14 		-94
16 		-93
18 		-95
20 		-97
25 		-95
30 		-95
40 		-97

Android device:
version: 4.4.4
build_number: KXB21.85-24 
model: Moto G Second Generation
manufacturer: Motorola

Beacon Info:
Kontact iBeacon
Advertisements per second: 10
Transmit Power: Max

IPhone 5s Average RSSI @1m: -51

Distance formula coefficients calculated for these values:
 Intercept: ?
 Multiplier: ?
 Power: ?

Bentleylust, 2016-03-10:
Anyone know how can I calculate coefficient1, coefficient2 and coefficient3?

I cannot find any approach tutorial for a*x^b + c. 
There are only methods for a*x^b. Would be grateful for any hint.

davidgyoung, 2016-03-10:
You can import the data into R (the free statistical computing software)
and use the nls function (non-linear least squares) to fit the data with a
power curve. Here is the power curve equation: function(x, b0, b1, b2) {b0
+ b1*(x^b2)}

The nls function solves for the coefficients b0, b1, and b2 which are used
in the formula for A, B, and C.

Here's a tutorial on curve fitting in R for more information:

http://www.walkingrandomly.com/?p=5254

closed 2016-03-14 by davidgyoung