AltBeacon/android-beacon-library #84: didRangeBeaconsInRegion finds zero beacons ...

mherrm, 2015-01-16:
Hello guys,

suddenly the function didRangeBeaconsInRegion does not return any beacons anymore. I use the libary in version 2.1 beta1. can u pls help ? its really annoying ....

thanks.

mherrm, 2015-01-16:
01-16 18:41:56.646  30017-30031/de.aaa.hms.app D/BeaconParser﹕ calculatedValue for position 1 with positionValue 256 and byteValue 0 is 0
01-16 18:41:56.647  30017-30031/de.aaa.hms.app D/BeaconService﹕ beacon detected multiple times in scan cycle :id1: f0018b9b-7509-4c31-a905-1a27d39c003c id2: 21568 id3: 6065
01-16 18:41:56.647  30017-30031/de.aaa.hms.app D/BeaconService﹕ beacon detected :id1: f0018b9b-7509-4c31-a905-1a27d39c003c id2: 21568 id3: 6065
01-16 18:41:56.648  30017-30031/de.aaa.hms.app D/BeaconService﹕ looking for ranging region matches for this beacon
01-16 18:41:56.648  30017-30031/de.aaa.hms.app D/BeaconService﹕ matches ranging region: id1: f0018b9b-7509-4c31-a905-1a27d39c003c id2: null id3: null
01-16 18:41:56.649  30017-30031/de.aaa.hms.app D/RangeState﹕ adding id1: f0018b9b-7509-4c31-a905-1a27d39c003c id2: 21568 id3: 6065 to existing range for: org.altbeacon.beacon.service.RangedBeacon@420c6470
01-16 18:41:56.650  30017-30031/de.aaa.hms.app D/BeaconParser﹕ This is not a matching Beacon advertisement.  (Was expecting be ac.  The bytes I see are: 0201061aff4c000215f0018b9b75094c31a9051a27d39c003c7adfcb88bf0000000000000000000000000000000000000000000000000000000000000000
01-16 18:41:56.650  30017-30031/de.aaa.hms.app D/BeaconParser﹕ This is a recognized beacon advertisement -- 0215 seen
01-16 18:41:56.651  30017-30031/de.aaa.hms.app D/BeaconParser﹕ Byte array is size 2
01-16 18:41:56.651  30017-30031/de.aaa.hms.app D/BeaconParser﹕ index is 0
01-16 18:41:56.651  30017-30031/de.aaa.hms.app D/BeaconParser﹕ calculatedValue for position 0 with positionValue 1 and byteValue 223 is 223
01-16 18:41:56.652  30017-30031/de.aaa.hms.app D/BeaconParser﹕ index is 1
01-16 18:41:56.652  30017-30031/de.aaa.hms.app D/BeaconParser﹕ calculatedValue for position 1 with positionValue 256 and byteValue 122 is 31232
01-16 18:41:56.652  30017-30031/de.aaa.hms.app D/BeaconParser﹕ Byte array is size 2
01-16 18:41:56.653  30017-30031/de.aaa.hms.app D/BeaconParser﹕ index is 0
01-16 18:41:56.653  30017-30031/de.aaa.hms.app D/BeaconParser﹕ calculatedValue for position 0 with positionValue 1 and byteValue 136 is 136
01-16 18:41:56.653  30017-30031/de.aaa.hms.app D/BeaconParser﹕ index is 1
01-16 18:41:56.653  30017-30031/de.aaa.hms.app D/BeaconParser﹕ calculatedValue for position 1 with positionValue 256 and byteValue 203 is 51968
01-16 18:41:56.654  30017-30031/de.aaa.hms.app D/BeaconParser﹕ Byte array is size 1
01-16 18:41:56.654  30017-30031/de.aaa.hms.app D/BeaconParser﹕ index is 0
01-16 18:41:56.654  30017-30031/de.aaa.hms.app D/BeaconParser﹕ calculatedValue for position 0 with positionValue 1 and byteValue 0 is 0
01-16 18:41:56.655  30017-30031/de.aaa.hms.app D/BeaconParser﹕ parsing found data field 0
01-16 18:41:56.655  30017-30031/de.aaa.hms.app D/BeaconParser﹕ Byte array is size 1
01-16 18:41:56.655  30017-30031/de.aaa.hms.app D/BeaconParser﹕ index is 0
01-16 18:41:56.655  30017-30031/de.aaa.hms.app D/BeaconParser﹕ calculatedValue for position 0 with positionValue 1 and byteValue 191 is 191
01-16 18:41:56.656  30017-30031/de.aaa.hms.app D/BeaconParser﹕ Byte array is size 2
01-16 18:41:56.656  30017-30031/de.aaa.hms.app D/BeaconParser﹕ index is 0
01-16 18:41:56.656  30017-30031/de.aaa.hms.app D/BeaconParser﹕ calculatedValue for position 0 with positionValue 1 and byteValue 21 is 21
01-16 18:41:56.656  30017-30031/de.aaa.hms.app D/BeaconParser﹕ index is 1
01-16 18:41:56.657  30017-30031/de.aaa.hms.app D/BeaconParser﹕ calculatedValue for position 1 with positionValue 256 and byteValue 2 is 512
01-16 18:41:56.657  30017-30031/de.aaa.hms.app D/BeaconParser﹕ Byte array is size 2
01-16 18:41:56.657  30017-30031/de.aaa.hms.app D/BeaconParser﹕ index is 0
01-16 18:41:56.657  30017-30031/de.aaa.hms.app D/BeaconParser﹕ calculatedValue for position 0 with positionValue 1 and byteValue 76 is 76
01-16 18:41:56.658  30017-30031/de.aaa.hms.app D/BeaconParser﹕ index is 1
01-16 18:41:56.658  30017-30031/de.aaa.hms.app D/BeaconParser﹕ calculatedValue for position 1 with positionValue 256 and byteValue 0 is 0
01-16 18:41:56.659  30017-30031/de.aaa.hms.app D/BeaconService﹕ beacon detected multiple times in scan cycle :id1: f0018b9b-7509-4c31-a905-1a27d39c003c id2: 31455 id3: 52104
01-16 18:41:56.660  30017-30031/de.aaa.hms.app D/BeaconService﹕ beacon detected :id1: f0018b9b-7509-4c31-a905-1a27d39c003c id2: 31455 id3: 52104
01-16 18:41:56.660  30017-30031/de.aaa.hms.app D/BeaconService﹕ looking for ranging region matches for this beacon
01-16 18:41:56.660  30017-30031/de.aaa.hms.app D/BeaconService﹕ matches ranging region: id1: f0018b9b-7509-4c31-a905-1a27d39c003c id2: null id3: null
01-16 18:41:56.661  30017-30031/de.aaa.hms.app D/RangeState﹕ adding id1: f0018b9b-7509-4c31-a905-1a27d39c003c id2: 31455 id3: 52104 to existing range for: org.altbeacon.beacon.service.RangedBeacon@420856c0
01-16 18:41:56.662  30017-30031/de.aaa.hms.app D/BeaconParser﹕ This is not a matching Beacon advertisement.  (Was expecting be ac.  The bytes I see are: 0201061aff4c000215f0018b9b75094c31a9051a27d39c003c544017b1bf0000000000000000000000000000000000000000000000000000000000000000
01-16 18:41:56.663  30017-30031/de.aaa.hms.app D/BeaconParser﹕ This is a recognized beacon advertisement -- 0215 seen
01-16 18:41:56.663  30017-30031/de.aaa.hms.app D/BeaconParser﹕ Byte array is size 2
01-16 18:41:56.664  30017-30031/de.aaa.hms.app D/BeaconParser﹕ index is 0
[... cut ...]
01-16 18:42:07.493  30017-31081/de.aaa.hms.app D/RangingData﹕ parsing RangingData
01-16 18:42:07.494  30017-31081/de.aaa.hms.app D/RangingData﹕ parsing RangingData
01-16 18:42:07.495  30017-31081/de.aaa.hms.app D/BeaconIntentProcessor﹕ got ranging data
01-16 18:42:07.495  30017-31081/de.aaa.hms.app D/ApplicationContextProvider﹕ didRangeBeaconsInRegion ... 0
01-16 18:42:07.499  30017-30029/de.aaa.hms.app D/CycledLeScanner﹕ got record
01-16 18:42:07.629  30017-30318/de.aaa.hms.app D/CycledLeScanner﹕ got record

davidgyoung, 2015-01-17:
Yes, that does sound frustrating.  I see that while bluetooth devices are detected by the CycledLeScanner, none are decoded by the BeaconParser class -- successfully or unsuccessfully after 18:42:04.  

Are you by chance making any calls to beaconManager.getBeaconParsers() that would then remove all of the beacon parsers?  Any other ideas on what this could be correlated with?  How long does it detect beacons successfully before this happens?  What do you need to do to get it to detect again?

Reviewing the library code, I'm not seeing anything else that would cause this.  You might also try the 2.1-beta2 release to see if anything changes with that.  The CycledLeScanner has changed quite a bit in that release, although I don't know of any changes that would fix this specific issue.

mherrm, 2015-01-17:
Thanks for you quick answer. I have to say that your libary worked very well since the last 2 days i started to face this issue without changing anything within the code ... 

Wether its successfull or not, it seems to be quite random .. i cannot find any specific pattern in it. 

btw: your download link for the beta2 regarding .aar file is not working :(

(https://github.com/AltBeacon/android-beacon-library/releases/download/2.1-beta2/android-beacon-library-2.1-beta2.aar)

mherrm, 2015-01-17:
I can add that it is not device dependent ... happens on moto g (2nd) and on samsung galaxy s4

quitequick, 2015-01-17:
FYI, I have seen this on devices using the 2.0 library. I thought it odd
but get around it with the code below. I just assumed it was honouring the
scanning frequency even if it found no beacons in the region.

       beaconManager.setRangeNotifier(new RangeNotifier() {
            @Override
            public void didRangeBeaconsInRegion(Collection<Beacon> beacons,
Region region)
            {
                if (beacons.size() == 0)
                    return;

mherrm, 2015-01-17:
Thank you. I already have that piece of code in my function. The result of getting 0 beacons is a huge delay in detecting them in my app ...

davidgyoung, 2015-01-17:
@mherrm, I have fixed the download link for 2.1-beta2 on the download page.  The correct link is https://github.com/AltBeacon/android-beacon-library/releases/download/untagged-7686360cf5d306bc0456/android-beacon-library-2.1-beta2.aar

One question to clarify, this statement: "The result of getting 0 beacons is a huge delay in detecting them in my app."  Are you saying that the detections resume after some period of time on their own?  If so, how long do they take to resume?

Since I have not been able to reproduce this yet, would you be willing to try running a custom snapshot build that adds extra logging to help identify the issue?

davidgyoung, 2015-01-17:
@mherm, does you application using the library use the `AsyncTask` class?  One possibility in what is happening here is that an `AsyncTask` execution is not completing for a long period of time, and as a result, it is blocking the execution of the `AsyncTask` internal to scanned beacon processing.

quitequick, 2015-01-17:
For my part, I see this zero size beacon list in didRangeBeaconsInRegion intermittently but generally always in keeping with the scanning frequency. At times this could be as much as every other call. So I upped the advertising frequency on the beacon and this happened to reduce it (even though the scan/between/periods should have caught them). That's why I thought I was getting scheduled calls to didRangeBeaconsInRegion despite zero found beacons. So I don't know if its because AltBeacon didn't find any beacons for the region due RF interference etc, or whether it did but somehow zero beacons are sent to didRangeBeaconsInRegion. Should didRangeBeaconsInRegion ever be called with zero beacons in the list?

But perhaps @mherrm is experiencing something different from me.

davidgyoung, 2015-01-17:
@quitequick, I think you are describing a very different issue.  The library is designed to call didRangeBeaconsInRegion at the defined interval regardless of whether any beacons have been seen in that period.  If no beacons were seen, then the length of the beacons list should be zero.  As you say, there are lots of reasons this might happen including the beacon transmit frequency, radio interference, etc.

The issue that @mherrm opened here is about the library suddenly stopping detections and repeatedly reporting seeing 0 beacons for many  ranging cycles even though beacons are transmitting in the vicinity.

mherrm, 2015-01-17:
Yes i can test a special snapshot :) i am currently checking on the asyntasks ...

mherrm, 2015-01-17:
@davidgyoung, i think u were right with the asynctasks ... i had a problem with the network connection and after fixing it it now works like a charm again :) thank you for your help. is this issue solveable? or is it an android operating system issue?

davidgyoung, 2015-01-17:
@mherrm, here is a link to the snapshot I prepared.  This is the same scanning code that is in 2.1-beta2, but I collect and log more debug information during the detection process.  Please run a build with this library version, and try to reproduce the problem.  If you can reproduce, I would like to see a logcat excerpt between two lines like below for a period where 0 beacons are being reported:

    01-16 18:42:06.414 30017-30028/de.aaa.hms.app D/CycledLeScanner﹕ got record

https://github.com/AltBeacon/android-beacon-library/releases/download/2.1-beta3-4-g7021c5a-SNAPSHOT/android-beacon-library-2.1-beta3-4-g7021c5a-SNAPSHOT.aar

mherrm, 2015-01-17:
When using your snapshot i get an build error regarding the manifest merger regarding the service exported value true with is in your libary now false :(

davidgyoung, 2015-01-17:
Sorry for the trouble!  I suspect that is happening because the AndroidManifest.xml for your app declares a conflicting entry for the BeaconService.  There is no reason to declare this service in your app's manifest because it will be inherited from the library.  I recommend removing it from your AndroidManifest.xml.  Alternatively, you can change the exported value in your manifest to be false so there is no merge conflict.

mherrm, 2015-01-17:
```
01-17 17:44:00.270  27053-27247/de.aaa.hms.app D/BeaconParser? calculatedValue for position 1 with positionValue 256 and byteValue 0 is 0
01-17 17:44:00.272  27053-27247/de.aaa.hms.app D/BeaconService? beacon detected multiple times in scan cycle :id1: f0018b9b-7509-4c31-a905-1a27d39c003c id2: 21568 id3: 6065
01-17 17:44:00.272  27053-27247/de.aaa.hms.app D/BeaconService? beacon detected :id1: f0018b9b-7509-4c31-a905-1a27d39c003c id2: 21568 id3: 6065
01-17 17:44:00.272  27053-27247/de.aaa.hms.app D/BeaconService? looking for ranging region matches for this beacon
01-17 17:44:00.272  27053-27247/de.aaa.hms.app D/BeaconService? matches ranging region: id1: f0018b9b-7509-4c31-a905-1a27d39c003c id2: null id3: null
01-17 17:44:00.273  27053-27247/de.aaa.hms.app D/RangeState? adding id1: f0018b9b-7509-4c31-a905-1a27d39c003c id2: 21568 id3: 6065 to existing range for: org.altbeacon.beacon.service.RangedBeacon@420246a0
01-17 17:44:00.273  27053-27247/de.aaa.hms.app D/BeaconParser? This is not a matching Beacon advertisement.  (Was expecting be ac.  The bytes I see are: 0201061aff4c000215f0018b9b75094c31a9051a27d39c003c7adfcb88bf0000000000000000000000000000000000000000000000000000000000000000
01-17 17:44:00.274  27053-27247/de.aaa.hms.app D/BeaconParser? This is a recognized beacon advertisement -- 0215 seen
01-17 17:44:00.274  27053-27247/de.aaa.hms.app D/BeaconParser? Byte array is size 2
01-17 17:44:00.275  27053-27247/de.aaa.hms.app D/BeaconParser? index is 0
01-17 17:44:00.275  27053-27247/de.aaa.hms.app D/BeaconParser? calculatedValue for position 0 with positionValue 1 and byteValue 223 is 223
01-17 17:44:00.275  27053-27247/de.aaa.hms.app D/BeaconParser? index is 1
01-17 17:44:00.275  27053-27247/de.aaa.hms.app D/BeaconParser? calculatedValue for position 1 with positionValue 256 and byteValue 122 is 31232
01-17 17:44:00.275  27053-27247/de.aaa.hms.app D/BeaconParser? Byte array is size 2
01-17 17:44:00.276  27053-27247/de.aaa.hms.app D/BeaconParser? index is 0
01-17 17:44:00.276  27053-27247/de.aaa.hms.app D/BeaconParser? calculatedValue for position 0 with positionValue 1 and byteValue 136 is 136
01-17 17:44:00.276  27053-27247/de.aaa.hms.app D/BeaconParser? index is 1
01-17 17:44:00.276  27053-27247/de.aaa.hms.app D/BeaconParser? calculatedValue for position 1 with positionValue 256 and byteValue 203 is 51968
01-17 17:44:00.277  27053-27247/de.aaa.hms.app D/BeaconParser? Byte array is size 1
01-17 17:44:00.277  27053-27247/de.aaa.hms.app D/BeaconParser? index is 0
01-17 17:44:00.277  27053-27247/de.aaa.hms.app D/BeaconParser? calculatedValue for position 0 with positionValue 1 and byteValue 0 is 0
01-17 17:44:00.277  27053-27247/de.aaa.hms.app D/BeaconParser? parsing found data field 0
01-17 17:44:00.277  27053-27247/de.aaa.hms.app D/BeaconParser? Byte array is size 1
01-17 17:44:00.277  27053-27247/de.aaa.hms.app D/BeaconParser? index is 0
01-17 17:44:00.277  27053-27247/de.aaa.hms.app D/BeaconParser? calculatedValue for position 0 with positionValue 1 and byteValue 191 is 191
01-17 17:44:00.278  27053-27247/de.aaa.hms.app D/BeaconParser? Byte array is size 2
01-17 17:44:00.278  27053-27247/de.aaa.hms.app D/BeaconParser? index is 0
01-17 17:44:00.278  27053-27247/de.aaa.hms.app D/BeaconParser? calculatedValue for position 0 with positionValue 1 and byteValue 21 is 21
01-17 17:44:00.278  27053-27247/de.aaa.hms.app D/BeaconParser? index is 1
01-17 17:44:00.278  27053-27247/de.aaa.hms.app D/BeaconParser? calculatedValue for position 1 with positionValue 256 and byteValue 2 is 512
01-17 17:44:00.279  27053-27247/de.aaa.hms.app D/BeaconParser? Byte array is size 2
01-17 17:44:00.279  27053-27247/de.aaa.hms.app D/BeaconParser? index is 0
01-17 17:44:00.279  27053-27247/de.aaa.hms.app D/BeaconParser? calculatedValue for position 0 with positionValue 1 and byteValue 76 is 76
01-17 17:44:00.279  27053-27247/de.aaa.hms.app D/BeaconParser? index is 1
01-17 17:44:00.279  27053-27247/de.aaa.hms.app D/BeaconParser? calculatedValue for position 1 with positionValue 256 and byteValue 0 is 0
01-17 17:44:00.281  27053-27247/de.aaa.hms.app D/BeaconService? beacon detected multiple times in scan cycle :id1: f0018b9b-7509-4c31-a905-1a27d39c003c id2: 31455 id3: 52104
01-17 17:44:00.281  27053-27247/de.aaa.hms.app D/BeaconService? beacon detected :id1: f0018b9b-7509-4c31-a905-1a27d39c003c id2: 31455 id3: 52104
01-17 17:44:00.281  27053-27247/de.aaa.hms.app D/BeaconService? looking for ranging region matches for this beacon
01-17 17:44:00.282  27053-27247/de.aaa.hms.app D/BeaconService? matches ranging region: id1: f0018b9b-7509-4c31-a905-1a27d39c003c id2: null id3: null
01-17 17:44:00.282  27053-27247/de.aaa.hms.app D/RangeState? adding id1: f0018b9b-7509-4c31-a905-1a27d39c003c id2: 31455 id3: 52104 to existing range for: org.altbeacon.beacon.service.RangedBeacon@4201e1a0
01-17 17:44:00.283  27053-27247/de.aaa.hms.app D/BeaconParser? This is not a matching Beacon advertisement.  (Was expecting be ac.  The bytes I see are: 0201061aff4c000215f0018b9b75094c31a9051a27d39c003c544017b1bf0000000000000000000000000000000000000000000000000000000000000000
01-17 17:44:00.283  27053-27247/de.aaa.hms.app D/BeaconParser? This is a recognized beacon advertisement -- 0215 seen
01-17 17:44:00.284  27053-27247/de.aaa.hms.app D/BeaconParser? Byte array is size 2
01-17 17:44:00.284  27053-27247/de.aaa.hms.app D/BeaconParser? index is 0
[... cut ...]
01-17 17:44:10.512  27053-27146/de.aaa.hms.app D/BeaconParser? calculatedValue for position 0 with positionValue 1 and byteValue 177 is 177
01-17 17:44:10.512  27053-27146/de.aaa.hms.app D/BeaconParser? index is 1
01-17 17:44:10.513  27053-27146/de.aaa.hms.app D/BeaconParser? calculatedValue for position 1 with positionValue 256 and byteValue 23 is 5888
01-17 17:44:10.513  27053-27146/de.aaa.hms.app D/BeaconParser? Byte array is size 1
01-17 17:44:10.513  27053-27146/de.aaa.hms.app D/BeaconParser? index is 0
01-17 17:44:10.513  27053-27146/de.aaa.

mherrm, 2015-01-17:
Within the didRangeBeaconsInRegion i am sometimes starting a new rest put on my server with an asyntask. removing this from the function i am getting a much cleaner beacon size almost all the time my two beacons ...

mherrm, 2015-01-17:
If i execute my Asyntask with executeOnExecuter(AsynTask.THREAD_POOL_Executor) instead of execute() it seems to be stable again ...

davidgyoung, 2015-01-17:
OK, that all makes sense.  The issue is that on recent Android versions, only one AsyncTask is allowed to run at a time in a single application.  Because beacon parsing in the Android Beacon Library is also done in an AsyncTask, if you have a long-running AsyncTask in your application it will block any further beacon detections until it is done.  

I think this is undesired behavior, so I think the library should be modified to use  `executeOnExecuter(AsyncTask.THREAD_POOL_Executor) ` for beacon parsing. 

I have put this into a new issue #88

Until this change is made, running your AsyncTask as `executeOnExecuter(AsyncTask.THREAD_POOL_Executor)` is a good workaround

mherrm, 2015-01-17:
Yeah thanks for your fast help :) good libary keeping going.

I am looking forward for android lollipop for my moto g 2nd regarding the battery consumption on scanning. in germany its still not out :(

davidgyoung, 2015-01-17:
thanks for your help in tracking this down, @mherrm!

closed 2015-01-17 by davidgyoung