AltBeacon/android-beacon-library #88: AsyncTask blocks beacon detections

davidgyoung, 2015-01-17:
On recent Android versions, only one AsyncTask is allowed to run at a time in a single application.  Because beacon parsing in the Android Beacon Library is also done in an AsyncTask, if you have a long-running AsyncTask in your application it will block any further beacon detections until it is done.  This problem was noticed in #84

I think this is undesired behavior, so I think the library should be modified to use  `executeOnExecuter(AsynTask.THREAD_POOL_Executor) ` for beacon parsing.

closed 2015-01-17 by davidgyoung

Yuhanlolo, 2018-05-21:
Hi, does this problem still exist? executeOnExecuter(AsynTask.THREAD_POOL_Executor) seems not work for me. The didRangeBeaconsInRegion keeps returning 0 for beacons.size(), but it used to work well. And my beacons are just nearby, which can also be detected by my computer.

davidgyoung, 2018-05-21:
No, I do not believe this problem exists since the modification was made over 3 years ago.  If you are using a recent library version, I suspect a different cause of the symptoms you are seeing.

commit 8aaaa6ca (2015-01-17): Fix #88 AsyncTask blocking beacon parsing