AltBeacon/android-beacon-library #95: bintray 1.0 publishing

davidgyoung, 2015-01-19:
When using the bintrayUpload task as described [here](https://bintray.com/jfrog/jfrog-jars/gradle-bintray-plugin/1.0/view/read), I get an error below.

@pieces029, any ideas?

g$ ./gradlew bintrayUpload -Prelease
[buildinfo] Properties file path was not found! (Relevant only for builds running on a CI Server)
:_bintrayRecordingCopy FAILED

FAILURE: Build failed with an exception.

* What went wrong:
A problem was found with the configuration of task ':_bintrayRecordingCopy'.
> No value has been specified for property 'destinationDir'.

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output.

BUILD FAILED

Total time: 5.125 secs

pieces029, 2015-01-19:
`./gradlew -Prelease release bintrayUpload` should do the trick.

davidgyoung, 2015-01-19:
I get the same result with that, too.

$ ./gradlew -Prelease release bintrayUpload
[buildinfo] Properties file path was not found! (Relevant only for builds running on a CI Server)
:clean
:androidJavadocsJar
:androidSourcesJar
:compileLint
:copyDebugLint UP-TO-DATE
:mergeDebugProguardFiles UP-TO-DATE
:preBuild
:preDebugBuild
:checkDebugManifest
:prepareDebugDependencies
:compileDebugAidl
:compileDebugRenderscript
:generateDebugBuildConfig
:generateDebugAssets UP-TO-DATE
:mergeDebugAssets
:generateDebugResValues UP-TO-DATE
:generateDebugResources
:packageDebugResources
:processDebugManifest
:processDebugResources
:generateDebugSources
:compileDebugJava
Note: /Users/dyoung/workspace/android-beacon-library/src/main/java/org/altbeacon/bluetooth/BluetoothCrashResolver.java uses or overrides a deprecated API.
Note: Recompile with -Xlint:deprecation for details.
Note: /Users/dyoung/workspace/android-beacon-library/src/main/java/org/altbeacon/beacon/service/BeaconService.java uses unchecked or unsafe operations.
Note: Recompile with -Xlint:unchecked for details.
:processDebugJavaRes
:packageDebugJar
:compileDebugNdk
:packageDebugJniLibs UP-TO-DATE
:packageDebugLocalJar UP-TO-DATE
:packageDebugRenderscript UP-TO-DATE
:bundleDebug
:assembleDebug
:copyReleaseLint UP-TO-DATE
:mergeReleaseProguardFiles UP-TO-DATE
:preReleaseBuild
:checkReleaseManifest
:prepareReleaseDependencies
:compileReleaseAidl
:compileReleaseRenderscript
:generateReleaseBuildConfig
:generateReleaseAssets UP-TO-DATE
:mergeReleaseAssets
:generateReleaseResValues UP-TO-DATE
:generateReleaseResources
:packageReleaseResources
:processReleaseManifest
:processReleaseResources
:generateReleaseSources
:compileReleaseJava
Note: /Users/dyoung/workspace/android-beacon-library/src/main/java/org/altbeacon/bluetooth/BluetoothCrashResolver.java uses or overrides a deprecated API.
Note: Recompile with -Xlint:deprecation for details.
Note: /Users/dyoung/workspace/android-beacon-library/src/main/java/org/altbeacon/beacon/service/BeaconService.java uses unchecked or unsafe operations.
Note: Recompile with -Xlint:unchecked for details.
:processReleaseJavaRes
:packageReleaseJar
:compileReleaseNdk
:packageReleaseJniLibs UP-TO-DATE
:packageReleaseLocalJar UP-TO-DATE
:packageReleaseRenderscript UP-TO-DATE
:bundleRelease
:assembleRelease
:assemble
:lint
Ran lint on variant release: 5 issues found
Ran lint on variant debug: 5 issues found
Wrote HTML report to file:/Users/dyoung/workspace/android-beacon-library/build/outputs/lint-results.html
Wrote XML report to /Users/dyoung/workspace/android-beacon-library/build/outputs/lint-results.xml
:preDebugTestBuild
:prepareComAndroidSupportSupportV42102Library
:prepareComSquareupFestAndroid108Library
:prepareDebugTestDependencies
:compileTestDebugJava
Note: /Users/dyoung/workspace/android-beacon-library/src/test/java/org/altbeacon/beacon/org/altbeacon/beacon/simulator/BeaconSimulatorTest.java uses or overrides a deprecated API.
Note: Recompile with -Xlint:deprecation for details.
:processTestDebugResources
:testDebugClasses
:testDebug
:test
:check
:build
:unzipAar
:bundleEclipse
:renameAarForRelease
:distribution
Building with version=2.1-beta-4
:release
Doing release build
:_bintrayRecordingCopy FAILED

FAILURE: Build failed with an exception.

* What went wrong:
A problem was found with the configuration of task ':_bintrayRecordingCopy'.
> No value has been specified for property 'destinationDir'.

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output.

BUILD FAILED

Total time: 19.074 secs
Davids-MacBook-Pro:android-beacon-library dyoung$

pieces029, 2015-01-19:
in the gradle/bintray.gradle can you try adding in the filesSpec closure `into "."` just after the from.

davidgyoung, 2015-01-19:
That worked.  I successfully published to Bintray, although the task still failed because Maven deploy part of it failed due to some unrelated issue.  I'll open a new issue about that.

closed 2015-01-19 by davidgyoung