AltBeacon/android-beacon-library #96: Maven publish fails through bintrayUpload

davidgyoung, 2015-01-19:
I got the error message below.  However, this did successfully publish to bintray, and I was then able to hit the dinosaur button in the web UI an the artifacts successfully sync'd to Sonotype OSS.

$ ./gradlew -Prelease release bintrayUpload
[buildinfo] Properties file path was not found! (Relevant only for builds running on a CI Server)
:clean
:androidJavadocsJar
:androidSourcesJar
:compileLint
:copyDebugLint UP-TO-DATE
:mergeDebugProguardFiles UP-TO-DATE
:preBuild
:preDebugBuild
:checkDebugManifest
:prepareDebugDependencies
:compileDebugAidl
:compileDebugRenderscript
:generateDebugBuildConfig
:generateDebugAssets UP-TO-DATE
:mergeDebugAssets
:generateDebugResValues UP-TO-DATE
:generateDebugResources
:packageDebugResources
:processDebugManifest
:processDebugResources
:generateDebugSources
:compileDebugJava
Note: /Users/dyoung/workspace/android-beacon-library/src/main/java/org/altbeacon/bluetooth/BluetoothCrashResolver.java uses or overrides a deprecated API.
Note: Recompile with -Xlint:deprecation for details.
Note: /Users/dyoung/workspace/android-beacon-library/src/main/java/org/altbeacon/beacon/service/BeaconService.java uses unchecked or unsafe operations.
Note: Recompile with -Xlint:unchecked for details.
:processDebugJavaRes
:packageDebugJar
:compileDebugNdk
:packageDebugJniLibs UP-TO-DATE
:packageDebugLocalJar UP-TO-DATE
:packageDebugRenderscript UP-TO-DATE
:bundleDebug
:assembleDebug
:copyReleaseLint UP-TO-DATE
:mergeReleaseProguardFiles UP-TO-DATE
:preReleaseBuild
:checkReleaseManifest
:prepareReleaseDependencies
:compileReleaseAidl
:compileReleaseRenderscript
:generateReleaseBuildConfig
:generateReleaseAssets UP-TO-DATE
:mergeReleaseAssets
:generateReleaseResValues UP-TO-DATE
:generateReleaseResources
:packageReleaseResources
:processReleaseManifest
:processReleaseResources
:generateReleaseSources
:compileReleaseJava
Note: /Users/dyoung/workspace/android-beacon-library/src/main/java/org/altbeacon/bluetooth/BluetoothCrashResolver.java uses or overrides a deprecated API.
Note: Recompile with -Xlint:deprecation for details.
Note: /Users/dyoung/workspace/android-beacon-library/src/main/java/org/altbeacon/beacon/service/BeaconService.java uses unchecked or unsafe operations.
Note: Recompile with -Xlint:unchecked for details.
:processReleaseJavaRes
:packageReleaseJar
:compileReleaseNdk
:packageReleaseJniLibs UP-TO-DATE
:packageReleaseLocalJar UP-TO-DATE
:packageReleaseRenderscript UP-TO-DATE
:bundleRelease
:assembleRelease
:assemble
:lint
Ran lint on variant release: 5 issues found
Ran lint on variant debug: 5 issues found
Wrote HTML report to file:/Users/dyoung/workspace/android-beacon-library/build/outputs/lint-results.html
Wrote XML report to /Users/dyoung/workspace/android-beacon-library/build/outputs/lint-results.xml
:preDebugTestBuild
:prepareComAndroidSupportSupportV42102Library
:prepareComSquareupFestAndroid108Library
:prepareDebugTestDependencies
:compileTestDebugJava
Note: /Users/dyoung/workspace/android-beacon-library/src/test/java/org/altbeacon/beacon/org/altbeacon/beacon/simulator/BeaconSimulatorTest.java uses or overrides a deprecated API.
Note: Recompile with -Xlint:deprecation for details.
:processTestDebugResources
:testDebugClasses
:testDebug
:test
:check
:build
:unzipAar
:bundleEclipse
:renameAarForRelease
:distribution
Building with version=2.1-beta-4
:release
Doing release build
:_bintrayRecordingCopy
:generatePomFileForDistPublication
:publishDistPublicationToMavenLocal
:bintrayUpload FAILED

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':bintrayUpload'.
> Could not sync 'altbeacon/android/android-beacon-library/2.1-beta-4' to Maven Central: HTTP/1.1 400 Bad Request [messages:[There are no published artifacts to sync for this version.], status:Validation Failed]

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output.

BUILD FAILED

Total time: 31.649 secs
Davids-MacBook-Pro:android-beacon-library dyoung$ git status
On branch master
Your branch is up-to-date with 'origin/master'.

pieces029, 2015-01-19:
Did you ever get the publishing to maven central worked out originally?

davidgyoung, 2015-01-19:
Troubleshooting it last week, I realized that there was no group for org.altbeacon (only for the old library package under com.radiusnetworks), so I created a new Jira ticket with OSS Sonatype on Friday, which got approved over the weekend.  This is the first time I tried it since.  

This is the very first time that hitting the dinosaur button in Bintray worked to publish to maven.  And it was with the same artifacts uploaded as described above.  It was just the automated publishing to Maven through the gradle task that failed.

davidgyoung, 2015-02-03:
The automation to Maven still does not work, but all it takes to get it there is to log in to Bintray and hit the Maven button, so this is very workable.

closed 2015-02-03 by davidgyoung

jjhesk, 2015-04-30:
Looks like i got the similar issue [from](https://github.com/novoda/bintray-release/issues/50) here

jjhesk, 2015-04-30:
maybe this will [help](https://github.com/AltBeacon/android-beacon-library/issues/96)