AltBeacon/android-beacon-library #97: Improve design of Identifier class

markvandertol, 2015-01-25:
I was looking at the Identifier class. I have some ideas to make the implementations of that class simpler. It also has a couple of bugs.

Most important improvements:

* It is confusing to have a single parse method that accepts all three forms of input, namely decimal, hexadecimal and uuid. I think clarity would be improved if there would be a specific one for each type and one generic that accepts a byte array.
* Make instances of this class immutable, the value no longer changes after it has been constructed, and therefore safe to share between threads. This is because other threads are unable to change the content of the instances after creation. This makes it also makes it safe to share this object with other objects without cloning, because it cannot change after it has been created. This makes it much easier to reason about concurrency and prevents bugs such as race conditions.

Some other smaller improvements:

* Strings aren’t the most efficient way to use as a normalized way to store the data. Byte arrays would be much simpler.
* Private no-param constructor isn’t needed when there is already another constructor.
* compareTo method follows the characteristics of the Java Comparable interface. Adding that interface makes it easier to use that method in other use cases.

Bugs I found:

* The Identifier constructor accepts null, but then most methods will throw null-pointer exceptions. I propose just to reject nulls in the constructor.
* Regular expressions in the class Identifier contain the following character classes: [0-9A-F-a-f]. The hyphen between the ‘F’ and the ‘a’ cause the ‘-’ to be an acceptable character. It will accept 2F234454-CF---4A0F-ADF2-F4911BA9FFA6 as an uuid.
* Only decimals are accepted with max length of 2 bytes. BeaconParser considers everything up to 4 bytes decimals (unsigned). Identifier can then return two different exceptions: Integer.parseInt() when it overflows or the check that the number is less than 65536. When a byte array is used to back the value, then the length of the identifier no longer matters.

If you also think that this is an improvement I could send in a pull request that refactors this class. The public API of this class stays compatible with the current version. What do you think?

davidgyoung, 2015-01-25:
Good ideas, @markvandertol.  The only one that worries me a bit is disallowing nulls.  I want to be careful this does not break backward compatibility.  An example of your proposal here might be useful.

Totally agree about the byte arrays, too. I have been meaning to do this for awhile, but have been put off by the effort to redactor the accompanying serialization in the Beacon class.

markvandertol, 2015-01-25:
I don't think an Identifier with a null value is currently in a working state, therefore I think it is better to fail early. That is directly when trying to create an instance of the object. Currently all public non-static methods, except toString, throw a NullPointerException when mStringValue is null. So currently there are a lot of places in the code that can cause a NullPointerException. Worse is that the code that triggers the NullPointerException can be far away from where the Identifier is instantiated, making debugging hard. Also what would an Identifier with a null value mean?

I shall create a pull request somewhere this week.

pieces029, 2015-01-25:
I agree, failing early is better.

davidgyoung, 2015-01-26:
@markvandertol, are you talking about the Identifier class or the Region class?  The Region class needs to accept nulls like in the reference app here:
https://github.com/AltBeacon/android-beacon-library-reference/blob/master/app/src/main/java/org/altbeacon/beaconreference/BeaconReferenceApplication.java#L46

markvandertol, 2015-01-26:
No, I'm talking about the Identifier class. The constructor allows null: https://github.com/AltBeacon/android-beacon-library/blob/d98e19f0064f21062f1bc76f40d5a3f872021e84/src/main/java/org/altbeacon/beacon/Identifier.java#L96 while the other methods of Identifier don't take into account that it could be null.

davidgyoung, 2015-01-26:
Gotcha.  Agreed.

davidgyoung, 2015-02-03:
Addressed in #103, thanks @markvandertol

closed 2015-02-03 by davidgyoung