AltBeacon/android-beacon-library #99: Monitoring and Ranging not working properly on low end devices

RaphaelRoyerRivard, 2015-01-27:
I am using your library v2.0 with Estimote beacons and it works very well on my Nexus 5. I monitor and do some ranging on over 100 beacons simultaneously and it is flawless. But on the **HTC Desire 320** (cheap but relatively recent phone that supports BLE), the library does not work like it does on my Nexus 5.
The didEnterRegion callcack is never called although the beacon is very close to the device (I tried with and without changing the default values of scanning periods). The didRangeBeaconsInRegion is called once in a while but always with 0 beacon detected.

I don't think that the Bluetooth module of the device is the problem, because use [iBeacon Detector](https://play.google.com/store/apps/details?id=youten.redo.ble.ibeacondetector&hl=en) to scan beacons and it works very well on that device...

Are you aware of this problem? If not, how can I help you finding the issue?

davidgyoung, 2015-01-27:
I would try an off the shelf app that also uses the library like Locate, and see if that works:  https://play.google.com/store/apps/details?id=com.radiusnetworks.locate&hl=en

RaphaelRoyerRivard, 2015-01-27:
Thank you, I will try that and give you the result tomorrow.

RaphaelRoyerRivard, 2015-01-27:
I had time to test it and it does not find the beacon that my Nexus 5 finds.

davidgyoung, 2015-01-27:
I would suggest running the reference app here:

https://github.com/AltBeacon/android-beacon-library-reference

With a modified `BeaconReferenceApplication` class that sets: `beaconManager.setDebug(true);`

Please capture a LogCat output and attach to this issue.

hugocroch, 2015-01-28:
I was reading about Beacon APIs and I found out that there a bug on the bluetoothLE Callback on some devices. An issue has been created : https://code.google.com/p/android/issues/detail?id=65863. Maybe it is the reason?

RaphaelRoyerRivard, 2015-01-28:
I do not have access to the HTC Desire 320 today, so I will send you the logs as soon as I can put my hand on it. In the meantime, maybe you will want to look at the [source of the iBeacon Detector app](https://github.com/youten/iBeaconDetector). It does not use your BeaconManager class to scan, so there is surely something it does differently.

davidgyoung, 2015-01-29:
The iBeacon Detector app uses the same scanning APIs as this library, and a copy of the IBeacon.java file from the 0.x version of the library to parse the scan results.   It sounds like there is something in the 2.x library that the HTC Desire 320 doesn't like, but it's hard to say what it is without seeing a debug log excerpt.  Please do send if you have a chance!

One more question, can you tell me the operating system version on the phone?  4.4?  5.0?

RaphaelRoyerRivard, 2015-01-29:
It is on 4.4.2, I hope I can have it today to retrieve the logs.

RaphaelRoyerRivard, 2015-01-30:
Actually, the reference app works on this phone, but not mine that uses your library... I'll search for what I did wrong that makes my app work on only some devices.

davidgyoung, 2015-02-03:
Hope you find the root cause, @RaphaelRoyerRivard.

closed 2015-02-03 by davidgyoung