AltBeacon/android-beacon-library #100: getBeaconParsers().add causes java.lang.UnsupportedOperationException

pionl, 2015-01-28:
Hi,

we are using beacon library and we had working code for searching custom beacons.

The code worked well, but recently we updated Android studio and gradle synced the beacon version to beta.

With this version the app throws java.lang.UnsupportedOperationException when using getBeaconParsers().add method.

Unfortunately the maven repository has only the latest version, so we can't revert the library version to 2.0.

Currently we are trying to create jar file from 2.0 version (we need to update the grade script).

Don't know if we are calling methods badly, or there is some bug.

Thank you for help

Code:

        mRegion = new Region("com.neogenia.spothill", null, null, null);
        mBeaconManager = BeaconManager.getInstanceForApplication(this);
        mBeaconManager.setBackgroundBetweenScanPeriod(1000l);
        mBeaconManager.setForegroundBetweenScanPeriod(1000l);

        mBeaconManager.getBeaconParsers().add(
                new BeaconParser().setBeaconLayout("m:0-3=4c000215,i:4-19,i:20-21,i:22-23,p:24-24"));
        mBeaconManager.debug = true;

        mBackgroundPowerSaver = new BackgroundPowerSaver(this);
        mRegionBootstrap = new RegionBootstrap(this, mRegion);

Stack trace:

            java.lang.UnsupportedOperationException
            at java.util.Collections$UnmodifiableCollection.add(Collections.java:928)
            at imakers.tools.MainService.onCreate(MainService.java:78)
            at android.app.ActivityThread.handleCreateService(ActivityThread.java:2736)

davidgyoung, 2015-01-28:
Sorry you are having trouble, @pioni.  Let me see if I can reproduce.  

Sorry, I had to redact part of your question as we cannot post the proprietary beacon layouts in this forum.

davidgyoung, 2015-01-28:
@pionl, sorry I cannot reproduce.  I tried using the reference app here:  https://github.com/AltBeacon/android-beacon-library-reference

And modified it to use the version 2.1-beta-4 version from Maven (by commenting out the flatDir section of the build.gradle file).  I also pasted your exact code above into the `onCreate` method of BeaconReferenceApplication.java.  I ran the code and everything executed successfully -- I did not get the same exception.

Are you absolutely positive it is that line that is throwing the exception?

Can you please verify your version?  To do that please add this line to your code:        `mBeaconManager.setDebug(true);` then look for a line in Logcat like this:

`I/BeaconService(25292): beaconService version 2.1-beta-4 is starting up`

quitequick, 2015-01-28:
I had an exception thrown today on the same method ...

mBeaconManager.getBeaconParsers().add(
            new BeaconParser().setBeaconLayout(myLayout));

It was because I had (accidentally) called that code twice with the same myLayout.  Calling it only once solved it. But it maybe a different thing altogether from @pionl's issue.

I'm using a beta of 2.1, fwiw.

davidgyoung, 2015-01-29:
@quitequick, I cannot reproduce an exception with the case you describe, either.  I tried:

        String layout = "m:2-3=aabb,i:4-19,i:20-21,i:22-23,p:24-24,d:25-25";
        mBeaconManager.getBeaconParsers().add(
                new BeaconParser().setBeaconLayout(layout));
        mBeaconManager.getBeaconParsers().add(
                new BeaconParser().setBeaconLayout(layout));

And I did not get an exception.  This is also with  2.1-beta-4.  Any further steps to reproduce would be appreciated.

pionl, 2015-01-29:
We had custom service, onCreate created manager. Downgrading the version did not help (it's strange that few weeks ago, it worked).

We managed to move the setup of the mBeaconManager.getBeaconParsers() to custom Application class and all is working correctly now.

Dont understand why it worked.

Thank you

closed 2015-01-29 by davidgyoung

peterbetos, 2015-02-03:
I was able to reproduce this. You need to find a way to instantiate the beaconManager when you close your app on the "Recent App" section (I was able to do this by triggering the manager via service).

BlueTrin, 2015-02-21:
I had the same issue:

http://stackoverflow.com/questions/28595468/exception-when-trying-to-add-a-beaconparser-to-altbeacon-lib

BlueTrin, 2015-02-21:
I think the issue seems to be that if you try to set the Layout in the class RangingActivty, something has happened before which is causing the trigger.

I found the commented code in BeaconReferenceApplication() 

        // By default the AndroidBeaconLibrary will only find AltBeacons.  If you wish to make it
        // find a different type of beacon, you must specify the byte layout for that beacon's
        // advertisement with a line like below.  The example shows how to find a beacon with the
        // same byte layout as AltBeacon but with a beaconTypeCode of 0xaabb.  To find the proper
        // layout expression for other beacon types, do a web search for "setBeaconLayout"
        // including the quotes.
        //

When I put the string there by uncommenting the code rather than  in RangingActivity, there was no error anymore.

I guess the object has a state where the layout cannot be changed anymore. Since this is the second time I am touching some kind of Java code, I am not really able to understand yet and help :)

Sorry !

furkanvarol, 2015-02-21:
I have checked the code, most probably you are adding new parser after binding beacon service (which forbids future custom layout adding).
```
public List<BeaconParser> getBeaconParsers() {
    if (isAnyConsumerBound()) {
        return Collections.unmodifiableList(beaconParsers);
    }
    return beaconParsers;
}
```
I don't know whether this will solve your problem but this can give you an idea to find what is missing in your project.

BTW, I think AltBeacon library should be updated to support adding custom parser at any time. Also it will eleminate problems like @peterbetos said.

BlueTrin, 2015-02-22:
So currently, how can I add a custom parser at a later time ? Do I need to unbind everything ?

furkanvarol, 2015-02-22:
Yes, it seems you have to unbind everything (which includes `RegionBootsrap`s).

daniel75kim, 2015-04-16:
I got the same issue as following, an exception : java.lang.UnsupportedOperationException

E/AndroidRuntime( 1956): Caused by: java.lang.UnsupportedOperationException
E/AndroidRuntime( 1956): at java.util.Collections$UnmodifiableCollection.add(Collections.java:928)

,when application hits the line of my implementation code
mBeaconManager.getBeaconParsers().add(new BeaconParser().
                setBeaconLayout("m:2-3=0215,i:4-19,i:20-21,i:22-23,p:24-24,d:25-25"));

It does not happen every time, in my case it only happens when the mobile boots up because it is running right after system booting, just after receiving the broad cast receiver (<action android:name="android.intent.action.BOOT_COMPLETED" />).

It happens 100% in my Moto G device (Android 4.4.4) when the device boots up without any USB cable connection after normal power off.  (frequency decreased quite a lot when I use the command (./adb reboot) or I connect the device with USB cable after normal power off) 

However, I was able to reproduce the situation that never happens this fatal error only if I use the library (versionName : "2.0", versionCode "3"), unlike  (versionName : "2.1.4", versionCode "1").

Once I used the version (versionName : "2.0", versionCode "3"), it never happens, 0% (over 50 times power off & power on).

I hope it will help you

davidgyoung, 2015-04-16:
To be clear, the core problem here is that **the library is not built to allow you to change the beaconParsers after it starts looking for beacons**, but unfortunately, the API is not designed to prevent this other than by throwing the java.lang.UnsupportedOperationException, which is admittedly confusing and inconvenient.

You can avoid this exception by taking care to **set your beaconParsers only once when your app starts up before you start ranging or monitoring**.  

I am open to ideas on how to lessen the pain of this until we can make a breaking change to the API in the next major release to better enforce the fact that you cannot add new beacon parsers after you start scanning.  (Allowing this creates many complexities with synchronization and region entry/exit states.)

davidgyoung, 2015-09-29:
Fixed in #290