AltBeacon/android-beacon-library #102: Regions enrolled as list in regionbootstrap only detects the last item on the monitor list

peterbetos, 2015-01-30:
If you try using a list on the following method:
````
regionBootstrap = new RegionBootstrap(this, regionList);
````
it seems that whatever is the last Region item on the regionList, it is the only one compared to the Beacon instance, and not with the rest of the list. 

Please check if the use of monitoredRegionState on the BeaconService class is still treated as a list, i.e examined each region one by one,  as it doesn't seem to be when using it on Galaxy S4. Thanks.

davidgyoung, 2015-01-31:
@peterbetos, I tried to reproduce this today, but I was unable to.   In my tests of the reference application with two regions set up like below, I *do* see that two regions are checked:

```
01-31 15:07:57.377  21605-21735/org.altbeacon.beaconreference D/BeaconService﹕ This region (id1: 74278bda-b644-4520-8f0c-720eaf059935 id2: null id3: null) does not match beacon: id1: ee23d392-60b6-4cae-91dc-7ca1d28f4198 id2: 1 id3: 1
01-31 15:07:57.377  21605-21735/org.altbeacon.beaconreference D/BeaconService﹕ This region (id1: 5a4bcfce-174e-4bac-a814-092e77f6b7e5 id2: null id3: null) does not match beacon: id1: ee23d392-60b6-4cae-91dc-7ca1d28f4198 id2: 1 id3: 1
```
```
        BeaconManager.setDebug(true);
        Region region1 = new Region("backgroundRegion1",
                Identifier.parse("5A4BCFCE-174E-4BAC-A814-092E77F6B7E5"), null, null);
        Region region2 = new Region("backgroundRegion2",
                Identifier.parse("74278BDA-B644-4520-8F0C-720EAF059935"), null, null);
        ArrayList<Region> regions = new ArrayList<Region>();
        regions.add(region1);
        regions.add(region2);
        regionBootstrap = new RegionBootstrap(this, regions);
```

However, in setting this up, I committed a common coding error that briefly did reproduce your problem.  I failed to keep my unique identifiers for each region unique (easy to do when you copy and paste.)  When I had the code like this at first, I reproduced your exact symptoms:

```
        Region region1 = new Region("backgroundRegion",
                Identifier.parse("5A4BCFCE-174E-4BAC-A814-092E77F6B7E5"), null, null);
        Region region2 = new Region("backgroundRegion",
                Identifier.parse("74278BDA-B644-4520-8F0C-720EAF059935"), null, null);
```

Note in the above that both regions have a uniqueId of "backgroundRegion", which will cause the second one to *replace* the first one when they are both added.  Please, take a look and see if you are doing the same thing, it is easy to do!

peterbetos, 2015-02-02:
Arrgh. The effects of drinking too much energy drinks! Found the culprit as well:

````
Region toAdd = new Region("myRangingUniqueId_" + locationList, Identifier.parse(uuidValue.trim()), null, null);
````

i'll try if the index I used (I have it on for loop) works on the multiple region detection, then I'll close this. Thanks.

peterbetos, 2015-02-02:
It working now. Thanks. Closing this.

closed 2015-02-02 by peterbetos