AltBeacon/android-beacon-library #105: Android 5 only: Crash on exit of app while BT off

mattwritescode, 2015-02-02:
For Android 5 devices, while Bluetooth is turned off, app crashes immediately after pressing the home button. This is replicable in the https://github.com/AltBeacon/android-beacon-library-reference.

E/AndroidRuntime﹕ FATAL EXCEPTION: main
    Process: com.radiusnetworks.testapp, PID: 18116
    java.lang.IllegalStateException: BT Adapter is not turned ON
            at android.bluetooth.le.BluetoothLeUtils.checkAdapterStateOn(BluetoothLeUtils.java:136)
            at android.bluetooth.le.BluetoothLeScanner.startScan(BluetoothLeScanner.java:106)
            at android.bluetooth.le.BluetoothLeScanner.startScan(BluetoothLeScanner.java:101)
            at org.altbeacon.beacon.service.scanner.CycledLeScannerForLollipop.deferScanIfNeeded(CycledLeScannerForLollipop.java:98)
            at org.altbeacon.beacon.service.scanner.CycledLeScanner.scanLeDevice(CycledLeScanner.java:163)
            at org.altbeacon.beacon.service.scanner.CycledLeScanner.finishScanCycle(CycledLeScanner.java:258)
            at org.altbeacon.beacon.service.scanner.CycledLeScanner.scheduleScanCycleStop(CycledLeScanner.java:231)
            at org.altbeacon.beacon.service.scanner.CycledLeScanner$1.run(CycledLeScanner.java:227)
            at android.os.Handler.handleCallback(Handler.java:739)
            at android.os.Handler.dispatchMessage(Handler.java:95)
            at android.os.Looper.loop(Looper.java:135)
            at android.app.ActivityThread.main(ActivityThread.java:5221)
            at java.lang.reflect.Method.invoke(Native Method)
            at java.lang.reflect.Method.invoke(Method.java:372)
            at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:899)
            at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:694)

davidgyoung, 2015-02-03:
Fixed in release 2.1.1

closed 2015-02-03 by davidgyoung

Tushar-CCI, 2015-02-06:
I think there is one instance of this issue still to fix, Line number 173 which causes crash sometimes, find the trace below

java.lang.IllegalStateException: BT Adapter is not turned ON
            at android.bluetooth.le.BluetoothLeUtils.checkAdapterStateOn(BluetoothLeUtils.java:136)
            at android.bluetooth.le.BluetoothLeScanner.stopScan(BluetoothLeScanner.java:144)
            at org.altbeacon.beacon.service.scanner.CycledLeScannerForLollipop.deferScanIfNeeded(CycledLeScannerForLollipop.java:173)
            at org.altbeacon.beacon.service.scanner.CycledLeScanner.scanLeDevice(CycledLeScanner.java:165)
            at org.altbeacon.beacon.service.scanner.CycledLeScannerForLollipop$1.run(CycledLeScannerForLollipop.java:155)
            at android.os.Handler.handleCallback(Handler.java:739)
            at android.os.Handler.dispatchMessage(Handler.java:95)
            at android.os.Looper.loop(Looper.java:135)
            at android.app.ActivityThread.main(ActivityThread.java:5312)
            at java.lang.reflect.Method.invoke(Native Method)
            at java.lang.reflect.Method.invoke(Method.java:372)
            at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:901)
            at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:696)

davidgyoung, 2015-02-07:
Nice catch!  That was a copy/paste error.

mattwritescode, 2015-02-08:
Wait, so we need to make another release of android beacon library now? I'm going to wait a week after it comes out before putting it into all the other libraries this time.

Sent from Matt's least favorite form of communication.

AlexLardschneider, 2016-08-16:
I am still experiencing the same issues in the AltBeacon 2.9

```
Fatal Exception: java.lang.IllegalStateException: BT Adapter is not turned ON
       at android.bluetooth.le.BluetoothLeUtils.checkAdapterStateOn(BluetoothLeUtils.java:136)
       at android.bluetooth.le.BluetoothLeScanner$1.handleMessage(BluetoothLeScanner.java:85)
       at android.os.Handler.dispatchMessage(Handler.java:102)
       at android.os.Looper.loop(Looper.java:168)
       at android.app.ActivityThread.main(ActivityThread.java:5821)
       at java.lang.reflect.Method.invoke(Method.java)
       at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:797)
       at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:687)
```

It seems like this issue only happens on HTC devices, as Crashlytics is not reporting crashes on any other devices except HTC.

davidgyoung, 2016-08-16:
Thanks for this report.  Catching this exception seems like a reasonable fix.

davidgyoung, 2016-08-17:
Please see #414 for further discussion on a related issue with HTC devices.  This issue had a different root cause and is closed.