AltBeacon/android-beacon-library #106: Android 5 performance

furkanvarol, 2015-02-04:
Recently we have updated one of test devices from 4.4.2 to 5.0 (LG G3). We realized that performance has dropped drastically. We even resetted phone to delete old files (factory reset) but it didn't seem like change anything. On the other hand, other test device which has 4.4.2 (Galaxy S5 G900H) still works great.

I want to explain how we are using your lib. We have 3 regions, all of them reads all beacon (no filter) but they act like a mode state. Also we have one RegionBootstrap. So,that RegionBootstrap has active regions, and whenever a beacon entered a region we are starting ranging. Key point in here, when we activate or deactivate a region, we are disabling old RegionBootstrap and creating a new one with active regions. I told this long story maybe this is the one causing performance issues :) but we don't think in that way since G3 was working normal before Lollipop and S5 still works normal. Also, performance is low when regions activity status staying same.

Other configurations:
Background Mode: always on
Scan Period         : 2000 ms
Scan Delay          : 200 ms

Sorry for a long story, maybe we just missed a conf for Lollipop, we couldn't figure it out. Any help will be appreciated. Also, let us know if anything we need to do or if we can help you to improve library.

davidgyoung, 2015-02-04:
@furkanvarol, a few questions:

1. What version of the library are you using?  (2.0, 2.1-beta2, 2.1.1, etc)
2. Can you please describe the performance issues in detail?  What symptoms do you experience?

If you are using a variant of the 2.1 library, can you please try to add the following line of code as soon as possible in your app (before you set up the RegionBootstrap), and see if it makes a difference:

    BeaconManager.setAndroidLScanningDisabled(true);

furkanvarol, 2015-02-04:
1. We are using the latest 2.1.1 and updating as soon as an update is availabe (I am checkin releases regularly)
2. RegionBootstrap shows that there are beacons in the Region (with no filter) and we are starting Ranging but we do not receive any beacon (only an empty collection) but after a while like 10 secs beacons are starting to come. We haven't faced this issue on 4.4.2 and still works normaly with 2.1.1 lib.

I will add the line of code as soon as possible but I can not do it right now because I live in GMT+2. If there is anything else you want to know feel free to ask. I will be happy to help.

davidgyoung, 2015-02-04:
Ok, so by "performance" you do not mean the app is using a high amount of CPU/memory resources, you mean detections take longer.

What mobile device model are you using when you see this?

Do you see the same delay when the app is in the foreground as when it is in the background?

furkanvarol, 2015-02-05:
Sorry for confusion, yes I ment beacon scan performance (or the time detection takes).

We are using two test device and the problematic one is LG G3 with Android 5.0 and AltBeacon Lib 2.1.2.

I had a chance to test both device today (second one is Galaxy S5 - G900H with Android 4.4.2 and AltBeacon Lib 2.1.2). Scenario was capturing a beacon and defining an object with captured beacon. So, RegionBootsrap has defined with a Region (no filter) and ranging has started as soon as a beacon has entered the area.

Total test time                       : 7000 ms
Beacon scan mode               : Foreground
Beacon Scan Period             : 2000ms
Beacon Scan Delay              : 200ms (time between two scan)
AndroidLScanningDisabled  : true
BackgroundPowerSaver       : Not initialized
Log Mode                             : Verbose (all logs)

Original configuration was different, I tried to disable some features to make test as much as clean and accurate.

I tried to analyze the data as much as I can, if you want I can send the logs.
Android 4.4.2 (S5) : first le record receive took ~100ms
Android 5.0.0 (G3) : first le record receive took ~5000ms (meantime lib started le scan 2 times but couldn't get any record
Android 4.4.2 (S5) : each turn ~7 beacon is found (total 3 turn)
Android 5.0.0 (G3) : each turn ~1 beacon is found (total 3 turn)
Android 4.4.2 (S5) : in 3 turn 23 beacon has found
Android 5.0.0 (G3) : in 3 turn only 1 beacon is found and event that beacon couldn't received by the listener

The key point here Android 4.4.2 has managed by CycledLeScannerForJellyBeanMr2 where Android 5.0.0 has managed  by BluetoothLeScanner and like I said before this wasn't an issue when G3 has JellyBean.

davidgyoung, 2015-02-05:
Logs would be helpful.  If you do not want to attach them to this issue,  you can send to david @radiusnetworks.com.

Can you try testing with this line and see if you get different results?

`BeaconManager.setAndroidLScanningDisabled(true);`

This line disables use of the new Android 5 scanning APIs.  I fear these are unreliable on the LG device.

furkanvarol, 2015-02-05:
I already tested with that line. I sent an email which includes logs.

davidgyoung, 2015-02-05:
@furkanvarol, I am still digging through the logs, but it appears that the LG device is not detecting bluetooth devices as expected regardless of whether the pre-Android 5 APIs are used to scan or the Android 5 APIs.  Just a hunch, but it may be related to the way the chip handles "connectable" advertisements under the firmware that shipped with its Android 5 ROM.  

What kind of beacon are you testing with?  Do you know its transmit frequency (1 Hz, 10Hz, etc.) and whether it sends a connectable advertisements?  Some bluetooth chips handle connectable advertisements differently, only reporting when they are seen once.  I'd be curious to know if you can test with a non-connectable advertisement (e.g. an iPhone transmitter or a Nexus 6/9 transmitter using the Locate app here:  https://play.google.com/store/apps/details?id=com.radiusnetworks.locate&hl=en

I would also be curious to hear if you have any success with detecting beacons reliably with the LG device using the same Locate app described above.  It may be the same result, but it would be good to eliminate any possible issue in your app code.

furkanvarol, 2015-02-05:
We haven't face any issue with pre-Lollipop LG G3 device.

We are working with HMSensor boards right now which advertise in Apple's iBeacon standards. I don't know their transmits frequency but I can check it out. BTW, they are connectable.

I have tested with our application currently but I will do more test with your app tomorrow as proposed and I will share results asap. If I can get an iPhone test device tomorrow, I will test in non-connectable mode as well.

furkanvarol, 2015-02-06:
I had a chance to test 4 device with your application named "Locate". Devices are HTC One (4.4.3), Galaxy S5 (4.4.2), LG G2 (4.4.2) and LG G3 (5.0.0). I will name HTC One, Galaxy S5 and LG G2 to the first three device from now on. BTW, I also used your app in iOS platform to transmit beacon. I used predefined Radius Network Beacon, id1 starts with 2f or something like that I don't remeber exactly but it was the first one in the list. Also, I want to say that I those test below relies on my observation skill since there is no log to look at it but I repeated all test at least 5 times.

Firstly, I have tested with connectable beacons in general scan mode:
* Performance of the first three device were same. When first round of scan was completed, I was be able to see ~10 beacons.
* Performance of the LG G3 was low. When first round of scan was completed, only ~2 beacon were found. Also, approximately 5 period was needed to see all beacons around.

Secondly, I have chosen one beacon as a test subject and tried to locate it:
* Performance of the first three device were still same. In ten seconds, devices were able to read the beacon ~15 times. Sometimes, the beacon got lost but found again in half a second. So, it seems like our beacon transmit frequency is around 500ms-1000ms.
* Performance of the LG G3 was still low and in this case pretty obvious. In ten seconds, G3 were able to read beacon ~3 times. When the beacon got lost, finding the beacon took ~4000ms.

Thirdly, I have tested with non-connectable beacon transmitted by iPhone 6 in again general scan mode:
* Performance of the first three device were same. When first round of scan was completed, transmitted beacon found firstly. It was always at the top (I haven't sorted beacons by their proximity).
* Performance of the LG G3 was normal. When first round of scan was completed, transmitted beacon found firstly. It was always at the top (I haven't sorted beacons by their proximity). So, like you said performance is better with non-connectable ones but I think not just because of it.

Fourthly, I have choosed the beacon transmitted by iPhone 6 as a test subject and tried to locate it:
* Performance of the first three device were still same. In ten seconds, devices were able to read the beacon ~20 times and beacon never got lost. So, it seems like the transmitted beacon transmit frequency is lower than 500ms.
* Performance of the LG G3 was good but not perfect. In ten seconds, device was able to read the beacon ~13 times and yet sometimes the beacon got lost. When the beacon got lost, finding the beacon took around ~500ms. So, LG G3 still is not perfect with non-connectable transmitted beacons.

Unfortunately, We only have one device with Lollipop. So, like you said it could be LG's problem but it can be Lollipop problem either. I will try to find and test with another device with Lollipop but I can not promise that since we don't have any other. And like I said before, interested part of the LG G3 is that when pre-Lollipop was installed nothing was wrong but now it has a serious performance issue.

I hope those test will be beneficial for you. Thanks.

davidgyoung, 2015-02-06:
Very comprehensive tests, thank you!

I would conclude based on these tests that this ROM image for this LG G3 has hobbled BLE scanning support.  It is probably related to the driver since it affects both the 5.x and 4.x BLE APIs under 5.0, but did not affect the 4.x APIs under 4.4.2.

For the record can you please post the build number under Settings -> About Phone?

furkanvarol, 2015-02-06:
Build Number : LRX21R.A1419313395

furkanvarol, 2015-02-09:
Today, we got an update on LG G3 and I want to say that all performance problems have gone! I didn't do any test like last time but even with a few basic test, it showed that problems have been fixed. Thanks for your cooperation. You can find build number below, in case of a need.

Working Build Number: LRX21R.A1422018487
Problematic Build Number: LRX21R.A1419313395

closed 2015-05-06 by davidgyoung