AltBeacon/android-beacon-library #107: Android 5 only: Crash on exit of app while BT off (NullPointerException)

Tushar-CCI, 2015-02-04:
java.lang.NullPointerException: Attempt to invoke virtual method 'void android.bluetooth.le.BluetoothLeScanner.startScan(java.util.List, android.bluetooth.le.ScanSettings, android.bluetooth.le.ScanCallback)' on a null object reference
            at org.altbeacon.beacon.service.scanner.CycledLeScannerForLollipop.deferScanIfNeeded(CycledLeScannerForLollipop.java:97)
            at org.altbeacon.beacon.service.scanner.CycledLeScanner.scanLeDevice(CycledLeScanner.java:165)
            at org.altbeacon.beacon.service.scanner.CycledLeScanner.finishScanCycle(CycledLeScanner.java:261)
            at org.altbeacon.beacon.service.scanner.CycledLeScanner.scheduleScanCycleStop(CycledLeScanner.java:234)
            at org.altbeacon.beacon.service.scanner.CycledLeScanner$1.run(CycledLeScanner.java:230)
            at android.os.Handler.handleCallback(Handler.java:739)
            at android.os.Handler.dispatchMessage(Handler.java:95)
            at android.os.Looper.loop(Looper.java:135)
            at android.app.ActivityThread.main(ActivityThread.java:5312)
            at java.lang.reflect.Method.invoke(Native Method)
            at java.lang.reflect.Method.invoke(Method.java:372)
            at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:901)
            at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:696)

furkanvarol, 2015-02-04:
Did you get latest release 2.1.1 ? I believe they have fixed this problem on it. Check issue #105

Tushar-CCI, 2015-02-04:
Yes I am using 2.1.1. They fixed for IllegalStateException but I am getting NullPointerException.

furkanvarol, 2015-02-04:
Hmm you are right. I am having a problem with disabling bluetooth too but on Android 4.4.2. Even, I disabled bluetooth I can still receive beacon. I am not sure if I sould create a new issue for this. It seems like there is a problem with disabling bluetooth while library is active

Tushar-CCI, 2015-02-04:
That would be a different issue I think, since there are separate CycledLeScanner class is used for Lollipop(5.0), which I think is cause of this issue.

davidgyoung, 2015-02-04:
I will investigate.

davidgyoung, 2015-02-04:
Fixed in release 2.1.2

closed 2015-02-04 by davidgyoung

mklimek, 2015-03-23:
I use 2.1.4 relaease and it still appears but only for emulator Lollipop devices.

```java.lang.NullPointerException: Attempt to invoke virtual method 'android.bluetooth.le.BluetoothLeScanner android.bluetooth.BluetoothAdapter.getBluetoothLeScanner()' on a null object reference
            at org.altbeacon.beacon.service.scanner.CycledLeScannerForLollipop.getScanner(CycledLeScannerForLollipop.java:219)
            at org.altbeacon.beacon.service.scanner.CycledLeScannerForLollipop.deferScanIfNeeded(CycledLeScannerForLollipop.java:99)
            at org.altbeacon.beacon.service.scanner.CycledLeScanner.scanLeDevice(CycledLeScanner.java:165)
            at org.altbeacon.beacon.service.scanner.CycledLeScanner.finishScanCycle(CycledLeScanner.java:261)
            at org.altbeacon.beacon.service.scanner.CycledLeScanner.scheduleScanCycleStop(CycledLeScanner.java:234)
            at org.altbeacon.beacon.service.scanner.CycledLeScanner$1.run(CycledLeScanner.java:230)
            at android.os.Handler.handleCallback(Handler.java:739)
            at android.os.Handler.dispatchMessage(Handler.java:95)
            at android.os.Looper.loop(Looper.java:135)
            at android.app.ActivityThread.main(ActivityThread.java:5221)
            at java.lang.reflect.Method.invoke(Native Method)```