AltBeacon/android-beacon-library #108: Disabling Bluetooth while Ranging is active

furkanvarol, 2015-02-04:
When ranging is active, disabling bluetooth does not seem to make a difference. I am still able get beacons but after killing app via app switcher returns bluetooth state to disabled (like it should be).

Android Ver          : 4.4.2
Phone Model        : SM-G900H (Galaxy S5)
AltBeacon Lib       : 2.1.1
Background Mode: always on
Scan Period          : 2000ms
Scan Delay           : 200ms

davidgyoung, 2015-02-04:
I wonder if disabling bluetooth is blocked if an app is doing a BLE scan (either on this phone's ROM image or on 4.4.2 as a whole).

furkanvarol, 2015-02-04:
Yes indicator disappears as soon as I disable Bluetooth. Also I was checking log (debug was enabled) and all logs continued to come like nothing happens until I kill the app via app switcher.