AltBeacon/android-beacon-library #113: Alternative RSSI Measurement

IgorZaus, 2015-02-09:
Hi, 

I suggest an alternative way of measure the RSSI value. Instead of measuring a plain moving average I'd suggest to calculate an auto regressive moving average. The advantages would include less memory consumption, faster, smoother and better control over the smoothness.

An very naive implementation would be as follows:

/*
 * This RangedBeacon calculates its rssi on base of an auto regressive moving average (ARMA)
 * It needs only the current value to do this; the general formula is  n(t) = n(t-1) - c * (n(t-1) - n(t))
 * where c is a coefficient, that denotes the smoothness - the lower the value, the smoother the average
 * Note: a smoother average needs longer to "settle down"
 */
public class ARMARangedBeacon extends RangedBeacon {

    private static final String TAG = "ARMARangedBeacon";
    //initially set to min value
    private int armaMeasurement = -100;

    public ARMARangedBeacon(Beacon beacon) {
        super(beacon);
    }

	public void addMeasurement(Integer rssi) {
        LogManager.d(TAG, "adding rssi: %s", rssi);
        setTracked(true);
        armaMeasurement = Double.valueOf(armaMeasurement - 0.1*(armaMeasurement - rssi)).intValue();
        LogManager.d(TAG, "armaMeasurement: %s", armaMeasurement);
	}

	public boolean noMeasurementsAvailable() {
		return false;
	}

	protected double calculateRunningAverage() {
		return armaMeasurement;

	}

Of course instantiation in RangeState.addBeacon() should be configurable and not just:

 mRangedBeacons.put(beacon, new ARMARangedBeacon(beacon));

davidgyoung, 2015-02-13:
I agree a more flexible and configurable filtering of RSSI is a good idea.  Perhaps it could be designed to have a RssiFilter interface, and the active RssiFilter could be set on the BeaconManager.  The RangedBeacon could use whatever RssiFilter is configured.

A couple of different filters could be built in -- RunningAverageRssiFilter (default) and AmraRssiFilter, each with configrable parameters.

davidgyoung, 2015-06-01:
This has been implemented in #113 and #114

closed 2015-06-01 by davidgyoung

samskiter, 2015-09-29:
Hey @IgorZaus would you mind explaining the difference between your ARMA filter and an IIR. I can't connect the dots between ArmaRssiFilter and the wikipedia explanation of an ARMA model: https://en.wikipedia.org/wiki/Autoregressive%E2%80%93moving-average_model

IgorZaus, 2015-09-29:
Hi Sam again, actually there is no real difference when you implement it recursively. I think it depends a little on where you’re using it. The basic difference equation, that is used to calculate one possible IIR is often called an ARMA representation of the IIR. When I used such equations back in the early 90s it was a good way to calculate a « soft » average without knowning the whole time series at a time… 

So, I would say, you could also name the class „IIR“…

samskiter, 2015-09-30:
Ah OK fair enough. I asked around a little and apparently this is an exponential filter. https://en.wikipedia.org/wiki/Exponential_smoothing