AltBeacon/android-beacon-library #114: Beacon Tracking Cache

IgorZaus, 2015-02-09:
I wonder, if it was a good idea to implement a tracking cache. If scanning period and beacon frequency is about the same (say, one second), it happens quite often, that a beacon disappears for one scan period. In real life this is rather unlikely, so I suggest to change 
RangeState.finalizeBeacons() 
to something like that:

if (rangedBeacon.getTrackingAge() > RangedBeacon.MAX_TRACKING_AGE)
   rangedBeacon.setTracked(false);

and in RangeBeacon.addMeasurement() 
[...]
lastTracked = System.currentTimeMillis();
[...]

and implement

public long getTrackingAge() {
        return System.currentTimeMillis() - lastTracked;
}

I don't know if there are better methods for that. Currently, I implemented that behaviour on the client side, when I get the callback from the BeaconService and it works as exected and gives better results in terms of stability.

davidgyoung, 2015-02-13:
Yes, I often implement this on my client code, too.  I agree it may be a good idea to build this in to the library.  Because some apps may need the current behavior, any tracking cache should have a configurable interval that can be set to 0.

Perhaps this can be combined with #113 as it touches the same code.  

@IgorZaus , any interested in putting these changes together and sending as a pull request?

IgorZaus, 2015-02-16:
Yes, give me a couple of days to prepare some sources!

IgorZaus, 2015-02-18:
Created a pull request - this is my very first contribution to Gitub - hope, I did everything ok....

davidgyoung, 2015-02-19:
Thanks, @IgorZaus.  Looks like you did the procedure correctly, thanks!  I need to review the changes and do some testing.  I will be out of the country for the next week, but I will try this out when I am back.

davidgyoung, 2015-06-01:
Implemented as described.

closed 2015-06-01 by davidgyoung

ersin-ertan, 2016-10-11:
Could some documentation be provided with this functionality. I'm not understanding:

`BeaconManager.setUseTrackingCache();`
`beaconManager.setMaxTrackingAge();`

- what it does
- for what situation you would want to modify these values. 
- what the implications of not using this is
- how do the two methods work together

davidgyoung, 2016-10-11:
The methods are intended to be used separately.  If  `setUseTrackingCache(true)` is called, then beacons will be tracked forever by Ranging APIs, and will never disappear from the callbacks.  

By default this is set to false, and they expire and are no longer returned in the ranging callback if they have not been seen in the number of milliseconds indicated by `setMaxTrackingAge(...)`, which defaults to 5000.