AltBeacon/android-beacon-library #115: Maven Publishing Issue

pieces029, 2015-02-09:
Hey @davidgyoung,

Just realized `publish = true` needs to be added to the bintray.gradle under publications. That will solve the issue of the maven central sync not happening.

davidgyoung, 2015-05-06:
like this?

    bintray {
        user = bintrayUsername
        key = bintrayKey
        publications = ['dist', 'publish = true']
    ...

pieces029, 2015-05-06:
Sorry for the confusion that wasn't worded very well. It just needs to be on the top level bintray object.

Like here https://github.com/pieces029/android-spock/blob/master/gradle/bintray.gradle

I just happen to have it directly below publications.

davidgyoung, 2015-05-06:
that makes more sense.  thanks!

davidgyoung, 2015-05-06:
Fixed in 401bd8064f4892f925fd55f109664126fb1c7546

closed 2015-05-06 by davidgyoung