AltBeacon/android-beacon-library #116: ConcurrentModificationException

furkanvarol, 2015-02-12:
Recently I started to have this exception. Sometimes it crashes the app and sometimes don't. I will try to investigate the problem.

```
E/AndroidRuntime﹕ FATAL EXCEPTION: main
    java.util.ConcurrentModificationException
            at java.util.HashMap$HashIterator.nextEntry(HashMap.java:806)
            at java.util.HashMap$KeyIterator.next(HashMap.java:833)
            at org.altbeacon.beacon.BeaconManager$1.onServiceConnected(BeaconManager.java:574)
            at android.app.LoadedApk$ServiceDispatcher.doConnected(LoadedApk.java:1114)
            at android.app.LoadedApk$ServiceDispatcher$RunConnection.run(LoadedApk.java:1131)
            at android.os.Handler.handleCallback(Handler.java:733)
            at android.os.Handler.dispatchMessage(Handler.java:95)
            at android.os.Looper.loop(Looper.java:136)
            at android.app.ActivityThread.main(ActivityThread.java:5579)
            at java.lang.reflect.Method.invokeNative(Native Method)
            at java.lang.reflect.Method.invoke(Method.java:515)
            at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1268)
            at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1084)
            at dalvik.system.NativeStart.main(Native Method)
```

mattwritescode, 2015-02-12:
@furkanvarol this is already fixed correct? Or is this somehow separate from https://github.com/AltBeacon/android-beacon-library/pull/117?

furkanvarol, 2015-02-12:
I mixed my work email in #117 so I push again in #118. If u got the particular error that wrote in first comment than yes, it is fixed.

You can check it out, it was a single line fix actually. I just removed Map's put method in a  foreach (BeaconManager's service connection handler).

closed 2015-02-12 by mattwritescode