AltBeacon/android-beacon-library #122: onScanResult() method won't be called after a long time scanning

BaoEnhui, 2015-02-16:
Hi, 
Thanks for sharing this awesome Library which really saved us a lot of time.
I am now using a Nexus 7(Android 4.4) and I have an issue while using this library.

It runs well for the first time that I launched the example app. However, after running about 30 minutes(sometimes several hours),  there isn't any beacon can be detected and I did nothing about the device.

From the log, I found that "onScanResult()" was not printed which should be keep printing while working.
And If I turned off bluetooth then turn it on again, it starts to work again.

Do you have any idea about this issue?

davidgyoung, 2015-02-22:
There is a known issue with the Nexus 7 where bluetooth and WiFi do not work properly at the same time.   (This is a general issue an not specific to beacon detection.)  

To see if this is the problem, try disabling WiFi and repeating your tests.

If this is the issue, I recommend you detect similtaneous WiFi and Bluetooth being on inside your app and warn users they can interfere with eachother if they have a Nexus 4, Nexus 7, Moto G or Moto X.

BaoEnhui, 2015-02-23:
Thanks for your reply.
I've tried it several times without turning WIFI on, but I still have the same issue:(

davidgyoung, 2015-03-11:
Can you turn on library debugging with `BeaconManager.setDebug(true);` and then let this run until it stops detecting, and capture a log excerpt for at least 10 seconds while the app is in the foreground without detecting beacons?

BaoEnhui, 2015-03-13:
Here's the log that captured from Nexus 7 (Lollipop) by running your reference example. Sorry I have upgrade my device to Lollipop an have the same issue.
03-10 12:04:00.982 D/CycledLeScanner(15430): Scan started
03-10 12:04:00.982 D/CycledLeScannerForLollipop(15430): got record
03-10 12:04:00.982 D/CycledLeScannerForLollipop(15430): got record
03-10 12:04:00.982 D/CycledLeScannerForLollipop(15430): got record
03-10 12:04:00.983 D/BeaconParser(15430): This is not a matching Beacon advertisement. (Was expecting be ac.  The bytes I see are: 0201060302041817fff90001afc0f0195aaf07c473929d0002eb13d5b77b3a060874696638301106a6da37dec19afc80944ad8a80262c2be020a00000000
03-10 12:04:00.983 D/BeaconParser(15430): This is not a matching Beacon advertisement. (Was expecting 02 15.  The bytes I see are: 0201060302041817fff90001afc0f0195aaf07c473929d0002eb13d5b77b3a060874696638301106a6da37dec19afc80944ad8a80262c2be020a00000000
03-10 12:04:00.983 D/BeaconParser(15430): This is not a matching Beacon advertisement. (Was expecting be ac.  The bytes I see are: 02010613ff4c000c0e0062798d5aedee1607159c025622000000000000000000000000000000000000000000000000000000000000000000000000000000
03-10 12:04:00.983 D/BeaconParser(15430): This is not a matching Beacon advertisement. (Was expecting 02 15.  The bytes I see are: 02010613ff4c000c0e0062798d5aedee1607159c025622000000000000000000000000000000000000000000000000000000000000000000000000000000
03-10 12:04:00.992 D/BeaconParser(15430): This is not a matching Beacon advertisement. (Was expecting be ac.  The bytes I see are: 02010606ff4c000a010000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
03-10 12:04:00.992 D/BeaconParser(15430): This is not a matching Beacon advertisement. (Was expecting 02 15.  The bytes I see are: 02010606ff4c000a010000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
03-10 12:04:01.021 E/bt-btm  (19797): new address: 30:8c:fb:2f:85:bf
03-10 12:04:01.022 E/bt-btm  (19797): new address: 30:8c:fb:2f:85:bf
03-10 12:04:01.022 D/BluetoothLeScanner(15430): onScanResult() - ScanResult{mDevice=30:8C:FB:2F:85:BF, mScanRecord=ScanRecord [mAdvertiseFlags=-1, mServiceUuids=[d2d3f8ef-9c99-4d9c-a2b3-91c85d44326c], mManufacturerSpecificData={35888=[-5, 47, -123, -65, 0, 50, 41, -59, -55]}, mServiceData={}, mTxPowerLevel=-2147483648, mDeviceName=null], mRssi=-89, mTimestampNanos=242021576447631}
03-10 12:04:01.022 D/CycledLeScannerForLollipop(15430): got record
03-10 12:04:01.023 D/BeaconParser(15430): This is not a matching Beacon advertisement. (Was expecting be ac.  The bytes I see are: 11076c32445dc891b3a29c4d999ceff8d3d20cff308cfb2f85bf003229c5c900000000000000000000000000000000000000000000000000000000000000
03-10 12:04:01.023 D/BeaconParser(15430): This is not a matching Beacon advertisement. (Was expecting 02 15.  The bytes I see are: 11076c32445dc891b3a29c4d999ceff8d3d20cff308cfb2f85bf003229c5c900000000000000000000000000000000000000000000000000000000000000
03-10 12:04:01.026 E/bt-btm  (19797): new address: 4d:6d:4c:89:fc:12
03-10 12:04:01.026 E/bt-btm  (19797): Random unmatch
03-10 12:04:01.027 E/bt-btm  (19797): new address: 4d:6d:4c:89:fc:12
03-10 12:04:01.027 E/bt-btm  (19797): Random unmatch
03-10 12:04:01.027 D/BluetoothLeScanner(15430): onScanResult() - ScanResult{mDevice=4D:6D:4C:89:FC:12, mScanRecord=ScanRecord [mAdvertiseFlags=6, mServiceUuids=null, mManufacturerSpecificData={76=[12, 14, 0, 98, 121, -115, 90, -19, -18, 22, 7, 21, -100, 2, 86, 34]}, mServiceData={}, mTxPowerLevel=-2147483648, mDeviceName=null], mRssi=-90, mTimestampNanos=242021581513549}
03-10 12:04:01.028 D/CycledLeScannerForLollipop(15430): got record
03-10 12:04:01.028 D/BeaconParser(15430): This is not a matching Beacon advertisement. (Was expecting be ac.  The bytes I see are: 02010613ff4c000c0e0062798d5aedee1607159c025622000000000000000000000000000000000000000000000000000000000000000000000000000000
03-10 12:04:01.028 D/BeaconParser(15430): This is not a matching Beacon advertisement. (Was expecting 02 15.  The bytes I see are: 02010613ff4c000c0e0062798d5aedee1607159c025622000000000000000000000000000000000000000000000000000000000000000000000000000000
03-10 12:04:01.059 E/bt-btm  (19797): new address: 4d:6d:4c:89:fc:12
03-10 12:04:01.059 E/bt-btm  (19797): Random unmatch
03-10 12:04:01.096 E/bt-btm  (19797): new address: 4d:6d:4c:89:fc:12
03-10 12:04:01.096 E/bt-btm  (19797): Random unmatch
03-10 12:04:01.131 E/bt-btm  (19797): new address: 38:37:11:aa:99:40
03-10 12:04:01.139 E/bt-btm  (19797): new address: 01:17:c5:3b:d5:d4
03-10 12:04:01.140 E/bt-btm  (19797): new address: 01:17:c5:3b:d5:d4
03-10 12:04:01.141 D/BluetoothLeScanner(15430): onScanResult() - ScanResult{mDevice=01:17:C5:3B:D5:D4, mScanRecord=ScanRecord [mAdvertiseFlags=6, mServiceUuids=[000080e7-0000-1000-8000-00805f9b34fb], mManufacturerSpecificData={76=[2, 21, 35, -96, 26, -16, 35, 42, 69, 24, -100, 14, 50, 63, -73, 115, -11, -17, 3, -106, 44, 118, -73]}, mServiceData={000080e7-0000-1000-8000-00805f9b34fb=[-80, 32, 44, 46, 27, -42, 108, 73, 3, -106, 44, 118, 59, -43, -44, 80, -1, -1, -1, 0, 0, 0, -1]}, mTxPowerLevel=-2147483648, mDeviceName=null], mRssi=-90, mTimestampNanos=242021694611695}
03-10 12:04:01.141 D/CycledLeScannerForLollipop(15430): got record
[... cut ...]
03-10 12:04:35.385 D/CycledLeScannerForLollipop(15430): starting scan in SCAN_MODE_LOW_LATENCY
03-10 12:04:35.388 D/BtGatt.GattService(19797): registerClient() - UUID=2ee2b80e-53d5-4952-b298-391b178e43b3
03-10 12:04:35.389 D/BtGatt.GattService(19797): onClientRegistered() - UUID=2ee2b80e-53d5-4952-b298-391b178e43b3, clientIf=5
03-10 12:04:35.389 D/BluetoothLeScanner(15430): onClientRegistered() - status=0 clientIf=5
03-10 12:04:35.389 D/BtGatt.GattService(19797): start scan with filters
03-10 12:04:35.390 D/CycledLeScanner(15430): Waiting to stop scan cycle for another 1100 milliseconds

davidgyoung, 2015-03-13:
Yes, it does seem that bluetooth detections stop after a period.  What kind of beacons are these?  Is it possible that some other config app on the tablet is making a GATT connection to the beacons and stopping them from transmitting?  Do you have a second device you could try to scan with when this one is failing to detect to verify the beacons are still transmitting?

Also, could you try the Radius Networks  Locate Beacon app at the same time as the reference app?  If Locate can detect after the reference app stops, it may indicate a library issue.  If neither can detect, it may indicate a hardware or driver issue.

BaoEnhui, 2015-03-13:
Yes, It might be a hardware or driver issue for Radius Networks Locate Beacon app also failed to detect any beacon nearby. And right after I turn off and  on bluetooth, both apps start to detect beacons. Do you know is there anyone else observed this issue in Nexus 7 or any other device?

davidgyoung, 2015-03-13:
This is the first report I have heard of this on the Nexus 7.  I have received lots of reports of problems like this on the Nexus 7 with WiFi enabled (there is a problem with simultaneous WiFi/Bluetooth operation in the hardware).  However, you report that you see this even with WiFi off.  That is a new one!

I wish you had a second Nexus 7 2013 Edition to test with to see if this is a device-specific issue, but I realize it is very unlikely you have more than one.

If you decided to upgrade your device to 5.0, it would be worth retesting.  If it starts working with 5.0, this may suggest a driver issue fixed in the upgrade.  If it still does not work with 5.0 it seems more likely to be a hardware flaw -- either on the individual device or on Nexus 7 2013 Edition hardware in general.

davidgyoung, 2015-06-01:
Closing because we cannot reproduce on a second device.  I suspect this is a device-specific problem.

closed 2015-06-01 by davidgyoung

lendylongli, 2017-04-14:
@davidgyoung @BaoEnhui I have the same problem，When I turned off & on bluetooth,it work again. My device is huiwei mate 8, Android 6.0

davidgyoung, 2017-04-14:
@lendylongli, if the problem is corrected by turning bluetooth off and back on, that suggests a low-level problem on the device either with the bluetooth chip itself of the low-level bluetooth stack such that it gets in a bad state that disallows BLE detections until it is restarted as you describe.  There may be nothing we can do about this, but if you can capture more information about what triggers it and can get some detailed logs that provide interesting details, it may be worth opening a new issue.

lendylongli, 2017-04-14:
@davidgyoung This is easy to reproduce in my device.I modified some of the code, it will be easy to reproduce. `BeaconService.java` 
--    `private ExecutorService mExecutor;//it  will be problematic, when i start & stop BeaconService many times, the thread pool will create many times, this can cause memory leaks`
++   `private static final ExecutorService mExecutor = Executors.newSingleThreadExecutor();`

start scan with this params
`ScanPeriod 200ms`
`BetweenScanPeriod 0ms`

davidgyoung, 2017-04-14:
@lendylongli , I am splitting this off to #496 so as not to hijack this thread.  Please place any additional comments there.