AltBeacon/android-beacon-library #124: model distance calculations FileNotFound on startup

mattwritescode, 2015-02-16:
This is occurring with android-beacon-library 2.1.3 when running it within the android-beacon-library-reference app. No crash occurs, but the error log is as follows:

02-16 12:50:00.087  24434-24434/org.altbeacon.beaconreference E/ModelSpecificDistanceCalculator﹕ Cannot open distance model file /data/data/org.altbeacon.beaconreference/files/model-distance-calculations.json
    java.io.FileNotFoundException: /data/data/org.altbeacon.beaconreference/files/model-distance-calculations.json: open failed: ENOENT (No such file or directory)
            at libcore.io.IoBridge.open(IoBridge.java:456)
            at java.io.FileInputStream.<init>(FileInputStream.java:76)
            at org.altbeacon.beacon.distance.ModelSpecificDistanceCalculator.loadModelMapFromFile(ModelSpecificDistanceCalculator.java:152)
            at org.altbeacon.beacon.distance.ModelSpecificDistanceCalculator.loadModelMap(ModelSpecificDistanceCalculator.java:132)
            at org.altbeacon.beacon.distance.ModelSpecificDistanceCalculator.<init>(ModelSpecificDistanceCalculator.java:72)
            at org.altbeacon.beacon.distance.ModelSpecificDistanceCalculator.<init>(ModelSpecificDistanceCalculator.java:62)
            at org.altbeacon.beacon.service.BeaconService.onCreate(BeaconService.java:205)
            at android.app.ActivityThread.handleCreateService(ActivityThread.java:2731)
            at android.app.ActivityThread.access$1800(ActivityThread.java:144)
            at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1361)
            at android.os.Handler.dispatchMessage(Handler.java:102)
            at android.os.Looper.loop(Looper.java:135)
            at android.app.ActivityThread.main(ActivityThread.java:5221)
            at java.lang.reflect.Method.invoke(Native Method)
            at java.lang.reflect.Method.invoke(Method.java:372)
            at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:899)
            at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:694)
     Caused by: android.system.ErrnoException: open failed: ENOENT (No such file or directory)
            at libcore.io.Posix.open(Native Method)
            at libcore.io.BlockGuardOs.open(BlockGuardOs.java:186)
            at libcore.io.IoBridge.open(IoBridge.java:442)
            at java.io.FileInputStream.<init>(FileInputStream.java:76)
            at org.altbeacon.beacon.distance.ModelSpecificDistanceCalculator.loadModelMapFromFile(ModelSpecificDistanceCalculator.java:152)
            at org.altbeacon.beacon.distance.ModelSpecificDistanceCalculator.loadModelMap(ModelSpecificDistanceCalculator.java:132)
            at org.altbeacon.beacon.distance.ModelSpecificDistanceCalculator.<init>(ModelSpecificDistanceCalculator.java:72)
            at org.altbeacon.beacon.distance.ModelSpecificDistanceCalculator.<init>(ModelSpecificDistanceCalculator.java:62)
            at org.altbeacon.beacon.service.BeaconService.onCreate(BeaconService.java:205)
            at android.app.ActivityThread.handleCreateService(ActivityThread.java:2731)
            at android.app.ActivityThread.access$1800(ActivityThread.java:144)
            at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1361)
            at android.os.Handler.dispatchMessage(Handler.java:102)
            at android.os.Looper.loop(Looper.java:135)
            at android.app.ActivityThread.main(ActivityThread.java:5221)
            at java.lang.reflect.Method.invoke(Native Method)
            at java.lang.reflect.Method.invoke(Method.java:372)
            at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:899)
            at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:694)

RobertoIDL, 2015-02-17:
I've got the same issue. Some how after installing a similar application from Google Play I don't see this warning any more.

davidgyoung, 2015-02-19:
This error is misleading and should be suppressed.  I believe it appears the first time the app starts up when the file has not yet been created (a fresh copy is first downloaded online and then it is saved).  After it is saved the first time, the error shoud not appear again.

mattwritescode, 2015-02-20:
I had the same experience as well. I'll make the change and create a PR.

closed 2015-02-22 by davidgyoung