AltBeacon/android-beacon-library #125: model-distance-calculation.json

logo17, 2015-02-16:
Can somebody help me with this problem? I don't know why I'm getting this error.

Cannot open distance model file /data/data/com.test.test/files/model-distance-calculations.json
    java.io.FileNotFoundException: /data/data/com.test.test/files/model-distance-calculations.json: open failed: ENOENT (No such file or directory)
            at libcore.io.IoBridge.open(IoBridge.java:409)
            at java.io.FileInputStream.<init>(FileInputStream.java:78)
            at org.altbeacon.beacon.distance.ModelSpecificDistanceCalculator.loadModelMapFromFile(ModelSpecificDistanceCalculator.java:152)
            at org.altbeacon.beacon.distance.ModelSpecificDistanceCalculator.loadModelMap(ModelSpecificDistanceCalculator.java:132)
            at org.altbeacon.beacon.distance.ModelSpecificDistanceCalculator.<init>(ModelSpecificDistanceCalculator.java:72)
            at org.altbeacon.beacon.distance.ModelSpecificDistanceCalculator.<init>(ModelSpecificDistanceCalculator.java:62)
            at org.altbeacon.beacon.service.BeaconService.onCreate(BeaconService.java:205)
            at android.app.ActivityThread.handleCreateService(ActivityThread.java:2596)
            at android.app.ActivityThread.access$1800(ActivityThread.java:144)
            at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1287)
            at android.os.Handler.dispatchMessage(Handler.java:102)
            at android.os.Looper.loop(Looper.java:136)
            at android.app.ActivityThread.main(ActivityThread.java:5146)
            at java.lang.reflect.Method.invokeNative(Native Method)
            at java.lang.reflect.Method.invoke(Method.java:515)
            at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:732)
            at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:566)
            at dalvik.system.NativeStart.main(Native Method)
     Caused by: libcore.io.ErrnoException: open failed: ENOENT (No such file or directory)
            at libcore.io.Posix.open(Native Method)
            at libcore.io.BlockGuardOs.open(BlockGuardOs.java:110)
            at libcore.io.IoBridge.open(IoBridge.java:393)
            at java.io.FileInputStream.<init>(FileInputStream.java:78)
            at org.altbeacon.beacon.distance.ModelSpecificDistanceCalculator.loadModelMapFromFile(ModelSpecificDistanceCalculator.java:152)
            at org.altbeacon.beacon.distance.ModelSpecificDistanceCalculator.loadModelMap(ModelSpecificDistanceCalculator.java:132)
            at org.altbeacon.beacon.distance.ModelSpecificDistanceCalculator.<init>(ModelSpecificDistanceCalculator.java:72)
            at org.altbeacon.beacon.distance.ModelSpecificDistanceCalculator.<init>(ModelSpecificDistanceCalculator.java:62)
            at org.altbeacon.beacon.service.BeaconService.onCreate(BeaconService.java:205)
            at android.app.ActivityThread.handleCreateService(ActivityThread.java:2596)
            at android.app.ActivityThread.access$1800(ActivityThread.java:144)
            at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1287)
            at android.os.Handler.dispatchMessage(Handler.java:102)
            at android.os.Looper.loop(Looper.java:136)
            at android.app.ActivityThread.main(ActivityThread.java:5146)
            at java.lang.reflect.Method.invokeNative(Native Method)
            at java.lang.reflect.Method.invoke(Method.java:515)
            at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:732)
            at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:566)
            at dalvik.system.NativeStart.main(Native Method)

mattwritescode, 2015-02-16:
As far as I can tell, it's not a big deal. But I created an issue for this exact thing less than one minute before you did... so this is a duplicate. Please see: https://github.com/AltBeacon/android-beacon-library/issues/124

closed 2015-02-16 by mattwritescode

logo17, 2015-02-16:
I've been having problems with this in Android 4.4.4 in a One Plus One device. After that, I'm getting this other error:
Fatal signal 6 (SIGABRT) at 0x0000031c (code=0), thread 14638, and a crash occurs.