AltBeacon/android-beacon-library #127: Frequent crash on Lollipop Nexus 9

NAYOSO, 2015-02-18:
Hello I'm countering this exception a lot on Lollipop Nexus 9, I'm using the android-beacon-library v 2.1.3. Any Idea how to solve this? Thanks!

```
java.util.concurrent.RejectedExecutionException: Task android.os.AsyncTask$3@10243ea1 rejected from java.util.concurrent.ThreadPoolExecutor@3c09e8c6[Running, pool size = 5, active threads = 5, queued tasks = 128, completed tasks = 154]
            at java.util.concurrent.ThreadPoolExecutor$AbortPolicy.rejectedExecution(ThreadPoolExecutor.java:2011)
            at java.util.concurrent.ThreadPoolExecutor.reject(ThreadPoolExecutor.java:793)
            at java.util.concurrent.ThreadPoolExecutor.execute(ThreadPoolExecutor.java:1339)
            at android.os.AsyncTask.executeOnExecutor(AsyncTask.java:590)
            at org.altbeacon.beacon.service.BeaconService$1.onLeScan(BeaconService.java:300)
            at org.altbeacon.beacon.service.scanner.CycledLeScannerForLollipop$2.onScanResult(CycledLeScannerForLollipop.java:234)
            at android.bluetooth.le.BluetoothLeScanner$BleScanCallbackWrapper$1.run(BluetoothLeScanner.java:330)
            at android.os.Handler.handleCallback(Handler.java:739)
            at android.os.Handler.dispatchMessage(Handler.java:95)
            at android.os.Looper.loop(Looper.java:135)
            at android.app.ActivityThread.main(ActivityThread.java:5221)
            at java.lang.reflect.Method.invoke(Native Method)
            at java.lang.reflect.Method.invoke(Method.java:372)
            at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:899)
            at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:694)
```

davidgyoung, 2015-02-19:
This indicates that the device is backing up when scanning for beacons.  Pleae check to see if the library´s reference application or the Locate Beacon app in the Google Play store run properly on your device.  I have tested them both on the Nexus 9 without seeing these exceptions.

If these apps work OK, the most likely explanation is a theading issue with custom code.  @NAYOSO, Is it possible that you have an app that has long running code on a ranging callback?

NAYOSO, 2015-02-20:
@davidgyoung Yes it works normally, and yes I have long running code on a ranging callback. I trigger a standard bluetooth LE scan and advertising from that callback. How to prevent it? spawn another thread?

davidgyoung, 2015-02-22:
Yes, this is most certainly the issue.   You should spawn a new thread for any long running actions in the ranging or monitoring callbacks.  Should be an easy fix!

closed 2015-02-22 by davidgyoung

niko-tenten, 2015-03-16:
@davidgyoung Hello I tried to spawn new thread but it's not helping and when I tried to comment all the code inside the ranging callback it still throw out the error. Do you have any other idea? My code worked really well with the Kitkat and JellyBean. I also encounter this on Nexus 5 but I did not encounter this problem with the Locate Beacon app.

davidgyoung, 2015-03-16:
Can you show the code in the ranging callback?  Something in that code must be executing very slowly.

NAYOSO, 2015-03-17:
@davidgyoung this my code, but even if I tried to comment all this code I still got the same error.

```
if (beacons.size() > 0) {
            Beacon beacon = beacons.iterator().next();
            final String beaconsID = beacon.getId3().toHexString();
            if (!storedBeacons.contains(beaconsID) && reportDataSource.isOpen()) {
                new Thread(new Runnable() {
                    @Override
                    public void run() {
                        storedBeacons.add(beaconsID);
                        ReportModel reportModel = new ReportModel();
                        reportModel.setType(getResources().getString(R.string
                                .report_type_peripheral));
                        reportModel.setPeripheralID(peripheralID);
                        reportModel.setAction(getResources().getString(R.string
                                .report_action_found));
                        reportModel.setTimestamp(System.currentTimeMillis() / 1000);
                        reportDataSource.createReport(reportModel);
                    }
                });
            }
            int newPositionState;
            if (beacon.getDistance() > 3.5) {
                newPositionState = STATE_FAR_FROM_BEACONS;
            } else {
                newPositionState = STATE_NEAR_FROM_BEACONS;
            }
            if (newPositionState != positionState) {
                final int finalNewPositionState = newPositionState;
                runOnUiThread(new Runnable() {
                    @Override
                    public void run() {
                        changeToolbarAndBluetoothUIState(positionState, finalNewPositionState);
                    }
                });
                if (newPositionState == STATE_FAR_FROM_BEACONS) {
                    stopScan();
                } else {
                    startScan();
                }
            }
            positionState = newPositionState;
        }

```

For a note I set the RangeNotifier my Application class and call this from there does it make any difference? I also tried to do the same thing like what you did in android-beacon-library-reference with implementing BeaconConsumer and start ranging after the BeaconService connected but I still facing the same problem.

This only happens on the Lollipop.

Thank You!

davidgyoung, 2015-03-17:
What happens if you set:

     mBeaconManager.setAndroidLScanningDisabled(true);

Before starting ranging?

NAYOSO, 2015-03-17:
@davidgyoung Still the same error.