AltBeacon/android-beacon-library #130: Late initialisation of RegionBootstrap

jchesterisidoro, 2015-02-20:
In the reference application, RegionBootstrap is initialised in a custom application class on it's onCreate method and of course, the application class is called before any activity is called.

Is there a way to initialise RegionBootstrap inside an activity? I already tried making a static variable of RegionBootstrap so i can call it in a different activity, but unfortunately, it doesn't work.

BeaconApplication.regionBootstrap = new RegionBootstrap((BootstrapNotifier) this.getApplication(), downloadedBeacons);

The Regions I needed to be initialised will come from a server, so initialisation of RegionBootstrap must not come from the application class.

furkanvarol, 2015-02-20:
You can use:
```
BeaconManager.setMonitorNotifier(MonitorNotifier);
BeaconManager.startMonitoringBeaconsInRegion(Region);
```
But do not forget. In order to use BeaconManager methods, you have to wait until BeaconService is connected.
All of class are under
```
org.altbeacon.beacon
```
namepace.

BTW, you should ask your question via StackOverflow.com. This forum is reserved for feature requests and reports of reproducible bugs that can be specifically documented.

closed 2015-02-22 by davidgyoung

jchesterisidoro, 2015-02-23:
Will those methods open the app from the background and monitor the beacons even after android kills the app same as stated here? 
`http://altbeacon.github.io/android-beacon-library/resume-after-terminate.html`