AltBeacon/android-beacon-library #131: exitRegion when running multiple apps

eckoj433, 2015-02-23:
I noticed today that when i run two apps at the same time which both uses the library, i get "random" exitRegion when the beacon is right next to my phone. It seem to work fine when i only have one app running, but when i have two it happends once every/every other minute. When i get the exitRegion event i get enterRegion again ~10-20 seconds after.

This occured on an LG G3 device with lollipop, using a kontakt beacon with 350ms frequency, and 5000ms scanperiod on the apps.

library version 2.1.2

I can not say with 100% certanity that this is not due to errors in my implementation, but i figured i would mention it here incase others have experienced the same problem, or someone is able to reproduce it.

furkanvarol, 2015-02-23:
I faced performance problem on LG G3. After updating G3 from 4.4.2 to 5.0 it became problematic but afterwards LG released another update and solved the performance issue. So my problem was a buggy LG Bluetooth driver maybe you are having the same problem too. However, there were only one app. I don't know whether you had the same issue but I thought that you should know about it. Please compare your build number with below:
Problematic Build Number: LRX21R.A1419313395
Working Build Number: LRX21R.A1422018487

eckoj433, 2015-02-24:
i have indeed experienced some hardware/firmware related buggs with the bluetooth on my device. The bluetooth crashes and restarts sometimes, and when this happends i do indeed get an exitEvent, but the bluetooth restarting is displayed very clearly in the log with warnings. The exitEvents i get when running two apps are not following a bluetooth crash, however it is still possible that it is related. I will try to get my hands on another device to replicate the problem on.

furkanvarol, 2015-02-24:
Actually, I wasn't talking about bluetooth crashes. When I was working on problematic firmware, bluetooth didn't crash at all but I got lots of exit region events because read performance was very poor and like I said it turnout that build was problematic. And again maybe it is not the case you are facing with but I think worth to checking it.

Btw, it can be a problem with new Android Lolipop BLE can api and for this case you can easily check with following code (this should be the first line of your beacon scanner initialization code):
```
BeaconManager.setAndroidLScanningDisabled(true);
```

eckoj433, 2015-02-24:
Tested on an S4 with 4.4.4 and it worked flawlessly, so you are probably right, it is most likely LG related or Lollipop related.  im running LRX21R.A1421650137 on my LG which is higher than your problematic build but lower than your working build, and im located in sweden so updates are a bit delayed here.

Thanks for your help, i will try to get my hands on another lollipop device to test aswell, and try disabling androidLScanning aswell.

furkanvarol, 2015-02-24:
Your test results are really similar to what I have got (#106). And like you said this was only a problem with LG G3 with 5.0 not with S5 with 4.4.2. However your build number looks older than mine that I got now. That can be the issue. Please let me know if you test with another Lollipop device. I will try to test performance on S5 with 5.0 when I will have a chance.

Also, you should test with https://play.google.com/store/apps/details?id=com.radiusnetworks.locate. This will allow you to determine whether something wrong with G3 or your code or even with the library.

furkanvarol, 2015-02-24:
I had a chance to test with two different S5, G900FQ with 5.0 and G900H with 4.4.2. I have tested with Radius Networks Locate application that I told you before. Their performance were pretty same and even G900FQ were a bit better. After that, I also tested with my application (my scan time is 2000ms) and results were also same.

I know you said that there were two applications so I did one more test. I opened my app in background (beacon scan activated) and Radius Network's Locate on foreground and started to test. Both of the phone scanned beacons successfully. Btw, I think Locate's scan interval is around 1100 ms but I am not sure since I am not the owner.

eckoj433, 2015-02-24:
tested with the app you suggested, and my kontakt beacon disappears from the list at the same time as i get exit event on my own app, then reappears at the same time as i get an enter event. 

when i run the same setup on the S4, the "locate" app enver lose the signal from my kontakt beacon and it remains in the list. Similarly when i force stop all processes of my own apps so that only the "locate" app is running, i get a consistent signal and it never dropps from the list. 

The problem only occurs on my LG when running at least two apps that are scanning for beacons, and it seems more and more likely that this problem is LG specifik.  I saw that you got your update on the LG around 15 days ago, and when lollipop rolled out we were about a month behind the rest of the world in scandinavia, so hopefully we will receive a build within 2 months that fixes these issues.

until then i will try to get my hands on some other devices to attempt to recreate the problem.

davidgyoung, 2015-02-25:
I'm going to close this issue, as the consensus seems to be that the issue is related to the LG build and not to the library itself.  @eckoj433, please do comment on this issue with the results of your attempts to reproduce the problem on other devices, and reopen if you can do so.

closed 2015-02-25 by davidgyoung