AltBeacon/android-beacon-library #132: Revert minSdk back to 4

davidgyoung, 2015-02-24:
A change in https://github.com/AltBeacon/android-beacon-library/pull/66 increased the minSdk from 4 to 7.  At the time, @pieces029 said he thought this was needed because the upgrade to com.squareup:fest-android:1.0 required this minSdk.  

However, it appears that this is not the case.  I can successfully run tests with the minSdk to 4.  @pieces029, is there any other reason you think setting the minSdk to 7 is required?

pieces029, 2015-02-24:
I believe it was for fest-android, which had another dependency that required it. If it still works with the minSdkversion 4 the maybe it was another dependency that has since been removed.

davidgyoung, 2018-03-06:
closing because the issue is obsolete.  The minSdk has increased to 14 due to subsequent changes.

closed 2018-03-06 by davidgyoung