AltBeacon/android-beacon-library #137: Starting an app in the background when BootstrapNotifier is not implemented in an Application Class

jchesterisidoro, 2015-03-04:
In the sample code from http://altbeacon.github.io/android-beacon-library/samples.html, BootstrapNotifier is implemented in an Application class, and the RegionBootstrap is initialised also in the onCreate of Application class. However, this initialisation is too early because Application class runs first before any Activity. You need to know the beacon's Identifiers beforehand to allow starting the app in the background.
What if the list of Identifiers are needed to be downloaded first after a login?
Is it possible to implement the BootstrapNotifier in a non-application or non-activity/service (ordinary class) and use initialise the RegionBootstrap from there?

davidgyoung, 2015-03-04:
Good points. For technical reasons, the `RegionBootstrap` must be constructed in an `Application` class because it works by auto launching the app (and therefore the BeaconService) on boot or power connect/disconnect.  

However, it is possible to do what you want.  From your Activity, you can save your beacon identifiers needed for constructing the `Region` to persistent storage.  Then in the `Application` you can load these identifiers back from persistent storage, and if present, construct a `Region` for the `RegionBootstrap` based on this info.

This is a common enough use case that it is probably worth adding code to the library that can do all this automatically.

jchesterisidoro, 2015-03-05:
Yes, please. How can I help to implement this?

davidgyoung, 2015-03-05:
If you would like to take a stab at coding it up, make a fork, then send a pull request.  I'd be happy to discuss design details in advance, too, if you need help getting started or want to bounce ideas around.  I think it should be pretty straightforward.

jchesterisidoro, 2015-03-11:
This method:
> However, it is possible to do what you want. From your Activity, you can save your beacon identifiers needed for constructing the Region to persistent storage. Then in the Application you can load [...]

works only when the phone is restarted and the application is swiped off from the recent list. When I forced kill the app, it doesn't work anymore.

davidgyoung, 2015-03-11:
If you force kill an app from the settings screen this is true.  In the other cases, however, it will work.

jchesterisidoro, 2015-03-11:
So this is a limitation when the BootstrapNotifier is not implemented in the application class?

davidgyoung, 2015-03-11:
Lots of details about this in the following issue:

https://github.com/AltBeacon/android-beacon-library/issues/44

Basically, on most devices, killing the app from the task switcher allows it to start up automatically to look for beacons in the background after reboot or power connect/disconnect (as the library tries to do.)  However, on all known devices, killing the app from the Settings screen (force kill) does not allow it to start up again without the user launching it manually.

jchesterisidoro, 2015-03-13:
It works with the reference app though. Phone restart, task switcher and force kill. But with the BootstrapNotifier that's not implemented in the Application class, it only works with the first two.

davidgyoung, 2015-03-13:
@jchesterisidoro, the results in your last comment are not consistent with my testing.   I see that with the reference app on a Nexus 5 (Android 4.4 or 5.0) only the first two actions cause the app to start scanning for beacons (restart phone, task switcher kill then power connect or disconnect), and the third action (force killing the app from settings) does not allow it to restart after reboot or after power connect or disconnect.  This is true even with RegionBootstrap/BootstrapNotifier is implemented in the ApplicationClass.

jchesterisidoro, 2015-03-16:
That's strange. I'm testing it with LG G2 Mini and Samsung Galaxy S4 both Android 4.4.2. I added these lines so I won't have to wait for the default scanning period.

`beaconManager.setBackgroundBetweenScanPeriod(01);
beaconManager.setBackgroundScanPeriod(1100l);'

davidgyoung, 2015-04-04:
Rolling this into #151

closed 2015-04-04 by davidgyoung