AltBeacon/android-beacon-library #138: Get continuous iBeacons info with Android AltBeacon Library

GITdanieljc, 2015-03-10:
This issue comes from this thread: http://stackoverflow.com/questions/28930926/get-continuous-ibeacons-info-with-android-altbeacon-library

The LogCat is the following: http://pastebin.com/DXxheZqW

The first (and unique) iBeacon detection occurs at line 67 and I get the log TAG at line 101. After that I don't get any iBeacon detection.

davidgyoung, 2015-03-10:
The stack trace tells me the following:

```
03-10 22:06:39.795 Scan (no beacons detected)
03-10 22:06:40.585 App moves to the background.  scanning paused
03-10 22:06:40.945 App moves to the foreground scanning resumed
03-10 22:06:41.945 Scan (beacon detected twice)
03-10 22:06:43.115 Scan (beacon detected once)
03-10 22:06:44.235 Scan (no beacons detected)
03-10 22:06:45.385 Scan (no beacons detected)
```
So a beacon was detected in two back-to-back scan cycles, then was not detected again.  

Can you confirm the device model, operating system version, and Android Beacon Library version you used for this test?  Also can you confirm whether WiFi was on or off during the test?

GITdanieljc, 2015-03-11:
- Device model: Vodafone Smart 4 turbo (http://www.gsmarena.com/vodafone_smart_4_turbo-6507.php) 
- Android 4.4.4
- Library version: 2.1.3
WiFi was on and connected to wifi spot during the test.

davidgyoung, 2015-03-11:
Quote from the original StackOverflow question from @GITdanieljc:  "That's it. In my app I get the Log.i log just once (first time Ibeacon is seen), whereas on the reference app (where monitoring and ranging are in separated activities) I get the log everytime I receive an Ibeacon message."

The troubling thing about this is that based on the log excerpt in this ticket, no low-level bluetooth packets are detected by the Android device after 03-10 22:06:44.235.  It's hard to understand how the Android Beacon Library (let alone changes in the user code using it) could cause this to happen.

davidgyoung, 2016-01-07:
closing due to inactivity

closed 2016-01-07 by davidgyoung