AltBeacon/android-beacon-library #139: BeaconManager#unbind throws IllegalArgumentException

hborders, 2015-03-19:
Sometimes, `Beaconmanager#unbind` throws an `IllegalArgumentException`:
```
Caused by: java.lang.IllegalArgumentException: Service not registered: org.altbeacon.beacon.BeaconManager$1@b639ead
1	android.app.LoadedApk.forgetServiceDispatcher	LoadedApk.java, line 1029
2	android.app.ContextImpl.unbindService	ContextImpl.java, line 1808
3	android.content.ContextWrapper.unbindService	ContextWrapper.java, line 551
4	com.example.BeaconUtils$BeaconConsumerImpl.unbindService	BeaconUtils.java, line 195
5	org.altbeacon.beacon.BeaconManager.unbind
```

I'm calling `BeaconManager#bind` and `BeaconManager#unbind` synchronously in a background thread. I wait for `BeaconConsumer#onBeaconServiceConnect` or a timeout, then I wait for `MonitorNotifier#didDetermineStateForRegion` or a timeout, I call `BeaconManager#unbind`. Occasionally, `BeaconManager#unbind` fails. Based on my review of your source. Two things could cause this:

`BeaconManager#bind`'s call to `BeaconConsumer#bindService` could fail. `BeaconManager#bind` doesn't return this success `boolean` to the caller, nor does it handle this failure in any way internally. If we later call `BeaconManager#unbind`, the call will throw the `IllegalArgumentException` described above.

`BeaconManager#beaconServiceConnection` could receive an `#onServiceDisconnected` callback after `BeaconManager#bind` is called. Since `BeaconManager#beaconServiceConnection` is already unbound, when we later call `BeaconManager#unbind`, the call will throw the `IllegalArgumentException` described above.

Please make `BeaconManager#bind` return a success `boolean`, and add a `onBeaconServiceDisconnect` method to `BeaconConsumer`.

Thanks!

davidgyoung, 2015-03-19:
Thanks for the feedback, these are good suggestions.

I imagine you can catch the exception in unbind for now, but I agree that this needs to be fixed.

X4mp, 2015-07-14:
I run in the same problem. I use a custom `BeaconScanner` class which implements `BeaconConsumer` and I implement all the methods. For `bindService` and `unbindService` i forward the call to `mContext#bindService(intent, serviceConnection, flags)`. When I call `BeaconManager.unbind()` the `IllegalArgumentException` is thrown.

Is there any fix for this right now? If not, do you have any suggestions on how to use BeaconConsumer in a custom class?

ram231, 2018-08-02:
just want to say Hi because I get an error to a different platform, although it is somewhat related. I'm using Flutter project, I created an issue to the guy created the package using altbeacons. then i get this error.

darshithedpara, 2020-05-02:
@ram231 Could you share your code snippet I am facing same issue in my flutter project.

ram231, 2020-05-02:
> @ram231 Could you share your code snippet I am facing same issue in my flutter project.

@darshithedpara . The project is dead. I think you should find another alternatives like a flutter BLE. My project is now in graveyard and can't update the packages anymore.

davidgyoung, 2020-05-02:
A few tips:

1.  The best way around this problem for now is to reduce the frequency in which you call bind/unbind.  Don't do it every time a screen appears or disappears or you increase the risk of unbind failures.  

2. You can catch the exception in unbind as a fail safe.

3. Anyone who wants to work on a pull request for an internal fix is encouraged to do so.  API changes must be backward compatible and cause minimal changes.

davidgyoung, 2021-08-06:
closed per #1046

closed 2021-08-06 by davidgyoung