AltBeacon/android-beacon-library #140: Null Pointer exception after executing regionBootstrap.disable() [Randomly]

jayshildave, 2015-03-26:
Null pointer exception is randomly raised after executing stop ranging (possibly after stop monitoring as well) when using the latest altBeacon trunk source code from Github 2.1.4

Its in link with issue #25 where it was being crashed in the same method.
Now the crash is on line no. 391 in method processBeaconFromScan from the asynctask.
The null pointer happens while reading state inside monitoredRegionState.

Currently we are synchronising reading rangedRegionState but monitoredRegionState is not given the same safeguard for multiple threaded operations.

davidgyoung, 2015-03-26:
Do you have a stack trace?  Which library version is this?

jayshildave, 2015-03-26:
The stack trace is 
java.lang.RuntimeException: An error occured
while executing doInBackground()
	at android.os.AsyncTask$3.done(AsyncTask.java:300)
	at java.util.concurrent.FutureTask.finishCompletion(FutureTask.java:355)
	at java.util.concurrent.FutureTask.setException(FutureTask.java:222)
	at java.util.concurrent.FutureTask.run(FutureTask.java:242)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1112)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:587)
	at java.lang.Thread.run(Thread.java:841)
Caused by: java.lang.NullPointerException
	at org.altbeacon.beacon.service.BeaconService.processBeaconFromScan(BeaconService.java:391)
	at org.altbeacon.beacon.service.BeaconService.access$400(BeaconService.java:66)
	at org.altbeacon.beacon.service.BeaconService$ScanProcessor.doInBackground(BeaconService.java:439)
	at org.altbeacon.beacon.service.BeaconService$ScanProcessor.doInBackground(BeaconService.java:422)
	at android.os.AsyncTask$2.call(AsyncTask.java:288)
	at java.util.concurrent.FutureTask.run(FutureTask.java:237)
	... 3 more

And I am compiling the latest source code from trunk.

closed 2015-04-04 by davidgyoung

jayshildave, 2015-04-04:
Hi David,

I posted the stack trace on the issue did you see it?

BR,
Jayshil

davidgyoung, 2015-04-04:
@jayshildave, yes I saw the stack trace.  I put in a fix in commit 0f7d5b7.  See #146