AltBeacon/android-beacon-library #143: getInstanceForApplication should be renames & is prone to context leaks

samskiter, 2015-03-30:
getInstanceForApplication or the BeaconManager class should probably call getApplicationContext on the application context to prevent leaks.

The suggested use for the BeaconManager is from an activity, like so:

`private BeaconManager beaconManager = BeaconManager.getInstanceForApplication(this);`

To prevent the activity being leaked, the beaconmanager should really make sure it calls `getApplicationContext` on the passed context.

Secondly `getInstanceForApplication` implies you should be passing an application to the method, simply `getInstance` might be better

davidgyoung, 2015-03-30:
Fair point.  I think the simplest solution is to change this line:

`client = new BeaconManager(context);`

To:

`client = new BeaconManager(context.getApplicationContext());`

samskiter, 2015-03-30:
Lgtm. Or put the getApplicationContext() into the constructor.

cupakromer, 2017-03-30:
Can this be closed now that #381 has been merged?

davidgyoung, 2017-03-31:
Fixed in #381

closed 2017-03-31 by cupakromer