AltBeacon/android-beacon-library #144: BeaconService never sets scanningEnabled true

samskiter, 2015-03-30:
scanningEnabled is never set to true, meaning after setting background mode, the service continues to scan

davidgyoung, 2015-03-30:
Good catch.  This bug was apparently introduced in the migration to Android L scanning APIs.

samskiter, 2015-06-18:
Is this now fixed?

davidgyoung, 2015-06-18:
Apologies, I have not looked at this ticket in awhile so I'm trying to refresh my memory.  I remember the issue, but looking at the CycledLeSanner.java code, I see that mScanningEnabled is set to true here:

https://github.com/AltBeacon/android-beacon-library/blob/master/src/main/java/org/altbeacon/beacon/service/scanner/CycledLeScanner.java#L131

And that code dates back to 2014-10-06.  Is this where you saw the issue?

samskiter, 2015-06-18:
No, i think it was here: https://github.com/AltBeacon/android-beacon-library/blob/2.1.4/src/main/java/org/altbeacon/beacon/service/BeaconService.java#L76

samskiter, 2015-06-18:
Ah, looks like you fixed here: https://github.com/AltBeacon/android-beacon-library/commit/49db318d501bca901341303e5162166a51c68310

closed 2015-06-18 by samskiter

davidgyoung, 2015-06-18:
Indeed!  It's been a busy few months.  Thanks for jogging my memory.