AltBeacon/android-beacon-library #145: Potentially bad assumptions about service binding/unbinding

samskiter, 2015-03-30:
When unbinding, I see the following comment:

`// If this is the last consumer to disconnect, the service will exit`
https://github.com/AltBeacon/android-beacon-library/blob/master/src/main/java/org/altbeacon/beacon/BeaconManager.java#L303

However only one beaconServiceConnection is ever made. Service binding/unbinding is actually ref counted against connection objects, so unbinding even once will destroy the service as you have bound with the same connected throughout.

Android source reference is here: http://grepcode.com/file/repository.grepcode.com/java/ext/com.google.android/android/4.4.2_r1/android/app/LoadedApk.java#LoadedApk.forgetServiceDispatcher%28android.content.Context%2Candroid.content.ServiceConnection%29

You might also find my SO post here about bound services useful: http://stackoverflow.com/a/24458955/1137254

davidgyoung, 2015-03-30:
I am surprised that we have not heard reports of people having problems with this.   Perhaps the solution is to make the library bind a single connection to the service from the application, do its own reference count internally, and unbind from the service when its reference count reaches zero.

samskiter, 2015-03-30:
You wouldn't see a problem until you tried to bind from two places
simultaneously, (e.g. from the application object AND from an activity) so
I wouldn't expect to see it as a popular bug.

Given that you currently do some reference counting, your solution would
probably be easiest to implement but redoing the ref counting seems a
little clunky. You could also have a look at robospice for a
service/activity/manager type pattern.

I currently must unbind on a config change to make sure the library doesn't
retain my activity. So one slight concern I had was that during a config
change you might cause the destruction/creation of a service unnecessarily.
In the example I posted you could maintain a bound service connection
across config changes.

I'm not absolutely sure of the service is actually recreated in that
situation, might be worth doing some logging (if any part of service life
cycle is on the main thread you could be lucky, but I doubt it).

Maybe decoupling the context and the consumer could prevent this or maybe
you could use some extra logic (such as whether ranging/monitoring is
enabled) to bind/unbind from the service.

samskiter, 2015-06-18:
I found some issues in the reference application that is in part related to binding/unbinding... https://github.com/AltBeacon/android-beacon-library-reference/issues/10

samskiter, 2015-06-18:
Update on this bug - I think it should actually be ok. a bit more digging around in getServiceDispatcher and forgetServiceDispatcher shows that the ref counting is done on two things: the service connection AND the calling context.

I snapped the following after getServiceDispatcher. The two bindings are from the RegionBootstrap (in the referenceApplication) and the RangingActivity.

mServices = {android.util.ArrayMap@3851} android.util.ArrayMap@3851, size = 2
mServices.keyAt(0) = {org.altbeacon.beaconreference.RangingActivity@3950} 
mServices.keyAt(1) = {org.altbeacon.beaconreference.BeaconReferenceApplication@3616} 
mServices.valueAt(0).keyAt(0) = {org.altbeacon.beacon.BeaconManager$1@3626} 
mServices.valueAt(0).valueAt(0) = {android.app.LoadedApk$ServiceDispatcher@4060} 
mServices.valueAt(1).keyAt(0) = {org.altbeacon.beacon.BeaconManager$1@3626} 
mServices.valueAt(1).valueAt(0) = {android.app.LoadedApk$ServiceDispatcher@3697} 

I think this is worth highlighting now as it could cause some subtle bugs down the road, firstly because use of the `RegionBootstrap` or similar will affect the service lifecycle. e.g. I just confirmed that without it, a rotate of RangingActivity will result in the BeaconService being destroyed and recreated. Secondly, because if anyone repeats the pattern used in `RegionBootstrap` to implement `bindService` (quoted below) they will certainly see strangeness in the bind/unbind ref counting.
```
public boolean bindService(Intent intent, ServiceConnection conn, int arg2) {
    return RegionBootstrap.this.application.getApplicationContext().bindService(intent, conn, arg2);
}
```

One option for preventing this kind of implementation is:

`public <T extends Context & BeaconConsumer> void bind(T consumer)`

davidgyoung, 2015-06-18:
I agree that this should be simplified to reduce the chance for subtle bugs.  I would like to redo the library APIs to get rid of binding by the end user and do it only internally.  The first time you start ranging or monitoring it would bind internally, and once you completely stop it would unbind.  

These are breaking changes, though, so these are things I'd like to put in a 3.0 release.

davidgyoung, 2021-08-06:
closed per #1046

closed 2021-08-06 by davidgyoung