AltBeacon/android-beacon-library #148: Callbacks are not made on the main thread, allowing state to be

samskiter, 2015-03-31:


davidgyoung, 2015-03-31:
I'm not sure I understand why not making callbacks on the main thread causes this issue.  This is by design, with the intention that the user needs to explicitly execute code on the main thread inside the callback.

I appreciate the attention, @samskiter.

samskiter, 2015-03-31:
It's a combination of calling on a background thread and having an activity as the receiver (albeit, indirectly via the anonymous inner class).

Consider the sequence of events:

1. beacon is found -> background callback on Activity A
2. Activity A calls dispatches to the main thread to make some UI change
3. Activity A undergoes a configuration change. Activity A is replaced on screen by Activity B. Activity A is still retained by the call made above in step 2
4. callback arrives at Activity A, which updates its UI. This is never seen by the user

If the UI change mentioned is to pop up a dialog or change any fragment, we will certainly see a crash unless commitAllowingStateLoss() was used (and the state will be lost, as per step 4)

One nice fact about config changes is that internally they happen as a single sequential block. Meaning that if you submit your callback to the UI thread you will definitely be one side of a config change or the other. This means you must also make sure you read your notifier on the main thread as the config change will likely involve changing the notifier.

Hope that clarifies?

davidgyoung, 2015-03-31:
One concern I have with this change is that if an app developer has a lot of code inside their callback (which they should not do anyway, but folks still do it), having it execute on the main thread might make the UI lag.   Pushing this change to app developers might require them to re-write their currently working code.

I understand that the current implementation allows the user to shoot themselves in the foot if they are getting callbacks inside an Activity and that Activity disappears.  Perhaps this should simply be solved by updating the reference app to demonstrate how to do this cleanly with the current implementation.

samskiter, 2015-03-31:
Hmm, the regression argument is important, you're right. The way to fix things up on the reference implementation would be use a retained fragment to hold the notifier and post to the main thread from there (where you then read the fragments "getActivity()" and pass the event through).

You could support different kinds of listeners/notifiers for the implementation and deprecate the old setNotifier endpoint... I think in general it would be great to have multiple notifiers/listeners, so implementing that at the same time might result in a change of method name, alerting existing devs to the change...