AltBeacon/android-beacon-library #154: Beacon scanning not restarted after app is killed due to memory pressue

davidgyoung, 2015-04-04:
If an app using the library is terminated due to memory pressure, it is not restarted automatically unless using the RegionBootsrap, and a power event is detected.

lpellis, 2015-04-04:
I was going to make a pull request but rather just comment here.  In CycledLeScanner you start the alarmManager with `System.currentTimeMillis()` where you should be using `SystemClock.elapsedRealtime()`, so the callback is never fired

davidgyoung, 2015-04-04:
I do see these getting fired in testing, just not after the app is killed due to memory pressure.

davidgyoung, 2015-04-05:
@lpellis, doing some quick searching, most examples I see of using AlarmManager use System.currentTimeMillis() to schedule an alarm not the time since boot.  And as I mentioned before, I have tested that this alarm does fire in some cases, so I do not think this is the core issue.

lpellis, 2015-04-05:
I had the exact issue where scanning was stopped after the app was killed from memory pressure, and this change fixed it.   Are you sure you are actually seeing the actual callbacks fired from the alarmmanager?

The current code is `alarmManager.set(AlarmManager.ELAPSED_REALTIME_WAKEUP, System.currentTimeMillis() + milliseconds, mWakeUpOperation);`

ie you are using ELAPSED_REALTIME_WAKEUP, which means you need to use  SystemClock.elapsedRealtime()
http://developer.android.com/reference/android/app/AlarmManager.html#ELAPSED_REALTIME_WAKEUP

If you want to use `System.currentTimeMillis()` then you need to use RTC_WAKEUP instead of ELAPSED_REALTIME_WAKEUP

davidgyoung, 2015-04-05:
Yeah, @lpellis  you are right.  I just retested, and changing to `alarmManager.set(AlarmManager.RTC_WAKEUP, System.currentTimeMillis() + milliseconds, mWakeUpOperation);` makes it fire both after the app is killed due to memory pressure and after to force close the app in task manager.  Don't know how this ever worked before.

One trick about making this change is that it will give you a second entry event after the app restarts if you are already in a beacon region.  This second event is a problem for things like the library's reference app, which actually launches a full screen activity on beacon detection.  The end result is that each time the app gets killed due to memory pressure, it relaunches itself into the foreground five minutes later!  This could be super annoying for real apps that do this.  One solution would be to implement a change suggested in #151 where the state of monitored regions are tracked in non-volatile storage so if the app is killed and restarts itself, you don't get a second entry event if you were already in the region before app restart.  We may need to roll this into the changes for #151.

closed 2015-06-01 by davidgyoung