AltBeacon/android-beacon-library #155: Code goes spaghetti if viewed with tab width != 4

flowlo, 2015-04-05:
I just attempted to read `Beacon.java` and `Region.java` but totally went nuts as soon as I encountered that mess of whitespace.

*Tabs vs. Spaces* is an endless discussion. TBH I don't want to bring it up. This issue is about consistency and valid for both models: I see that most code uses 4 spaces as one level of indentation, but every now and then tabs kick in, which is suboptimal.

I personally choose tab width dependent on the language I read, and using a tab width of 2 here (really anything else than 4) will freak the reader out.

[image]

This should be normalised throughout the whole codebase and have tests to cover future work.

davidgyoung, 2015-04-05:
Yes, we should be standardized on a tab length of four and use spaces instead of the tab character.  This is following the Android code style guidelines.

flowlo, 2015-04-15:
@samskiter Feel free to have a look at #167

davidgyoung, 2015-05-06:
Fixed in #167

closed 2015-05-06 by davidgyoung