AltBeacon/android-beacon-library #158: How to disable logs from android beacon library ?

daniel75kim, 2015-04-07:
I am now using android beacon library, version 2.1.4.

How am i able to stop logs from displaying, for example the following logs. 

BluetoothAdapter( 2950): startLeScan(): null
BluetoothAdapter( 2950): onClientRegistered() - status=0 clientIf=5
BluetoothAdapter( 2950): startLeScan(): null

Is there any API that toggles logs of beacon library 'on' or 'off'?

Thanks

davidgyoung, 2015-04-07:
Unfortunately, I know of no way to turn these off.  Those logs are put out by the Android operating system's built-in BluetoothAdapter.  You will get them whenever you are doing a Bluetooth LE scan.

furkanvarol, 2015-04-07:
You can use:
```
LogManager(Loggers.EMPTY_LOGGER);
```
under `org.altbeacon.beacon.logging` package but this will turn of all AltBeacon logs.

If you want you can create a custom `Logger` and set it here so that you can manage which log you want to see from AltBeacon Lib.

daniel75kim, 2015-04-07:
Hi David,

Thank you for your reply.

Regards,
Daniel

>
> Unfortunately, I know of no way to turn these off. Those logs are put out
by the Android operating system's built-in BluetoothAdapter. You will get
them whenever you are doing a Bluetooth LE scan.

davidgyoung, 2015-04-07:
The problem in this case is that a logging put out by `BluetoothAdapter` comes from the Android operating system, so it doesn't respect the settings of the library's LogManager.  If you look at the source code for `BluetoothAdapter` https://android.googlesource.com/platform/frameworks/base/+/master/core/java/android/bluetooth/BluetoothAdapter.java you see a line like this:

        if (DBG) Log.d(TAG, "startLeScan(): " + serviceUuids);

So the only way to turn off that log line is to set DBG to false.  And DBG is defined in the same class like this:

       private static final boolean DBG = true;

I tried to use a semi-evil nuclear option to kill this like so:

        try {
            java.lang.reflect.Field field = android.bluetooth.BluetoothAdapter.class.getDeclaredField("DBG");
            field.setAccessible(true);
            field.set(null, false);
            Log.i(TAG, "Disabled logging from BluetoothAdapter");
        }
        catch (Exception e) {
            Log.e(TAG, "Cannot disable debug logging from BluetoothAdapter ", e);
        }

But this didn't seem to work.  Not sure why.

nick-at-humin, 2015-04-11:
This "nuclear" option doesn't work because the actual adapter is on the other side of a service connection.

davidgyoung, 2015-05-06:
The above nuclear option did not work because the logs run in a different process from the app, so you can't change the DBG flag.  Sorry, the only way to fix this is with an OS update by AOSP.

closed 2015-05-06 by davidgyoung