AltBeacon/android-beacon-library #161: Monitored/Ranged regions are lost after unbind/bind

samskiter, 2015-04-07:
After setting up some regions, unbinding and then binding again, I find that those regions are no longer monitored. I suspect that is because the beaconManager should re-send the regions it wants monitored after getting a new serviceMessenger, here:

https://github.com/AltBeacon/android-beacon-library/blob/master/src/main/java/org/altbeacon/beacon/BeaconManager.java#L581

JakobMaaloe, 2018-10-30:
I'm experiencing the same problem, after unbinding, rebinding doesn't give any scan results... 
Doesn't seem to be related to the time I wait after scanning.
What's the status of this bug?

davidgyoung, 2018-10-30:
@JakobMaaloe, can you please help create a reproducible test case starting with the [Reference Application](https://github.com/AltBeacon/android-beacon-library)?  I understand that requires a little effort, but please understand that this is a on open source project that relies on volunteers to help solve issues like these.  Ideally, you'd show any changes to the reference app needed to reproduce the issue, along with a log excerpt showing the unexpected behavior.

davidgyoung, 2021-08-06:
Closed in favor of #1046

closed 2021-08-06 by davidgyoung