AltBeacon/android-beacon-library #163: didRangeBeaconsInRegion finds zero beacons part 2

hippiefahrzeug, 2015-04-11:
Similarly as in issue #84 and this stackoverflow question [here](http://stackoverflow.com/questions/23830875/radiusnetworks-ibeacon-didrangebeaconsinregion-return-0-beacons), in my app didRangeBeaconsInRegion gets called often with 0 beacons found. Unfortunately my case is a bit stranger. This is what happens...

I use the library with default settings, configured with a beacon parser which detects ibeacons ("m:2-3=0215,i:4-19,i:20-21,i:22-23,p:24-24,d:25-25"). Monitoring always works beautifully, but ranging shows the following behavior:

- On an HTC One with Android 5.0.1: didRangeBeaconsInRegion gets called almost always with the beacon next to it. (I always have the beacon next to the device in these tests) -> great!
- On a Moto X 2nd gen (Android 5.0): didRangeBeaconsInRegion gets called with 0 beacons 9 out of 10 times. It's very erratic, sometimes (but rarely) it works ok for a couple of times. I added code which just uses the last detected beacon as long as I've not exited the region. But detection takes a very long time.

Normally I would say that there is either something fishy with the Moto X or the installed OS, but when I use the very old android library from two years ago (the old one which was pulled), it works very reliably on all devices.

I don't think I have the AsyncTask problem here either I also tried fiddling with the DEFAULT_FOREGROUND_SCAN_PERIOD value to no avail.

davidgyoung, 2015-04-11:
@hippiefahrzeug, on the Moto X, can you please try two things:

1. Set `beaconManager.setAndroidLScanningDisabled(true)` and see if it makes any difference.

2. Set beaconManager.setDebug(true);`  then capture a LogCat excerpt over several minutes while scanning in the foreground.  The log will be long, so you nay wish to put it in a Gist and link to it from here.

Apologies, but I had to remove your beacon parser expression in the issue above as we must keep intellectual property out of this open source repo.

davidgyoung, 2015-04-11:
One other thing to try: please test with the Locate Beacon app, which is based on this library, and see if you also get missed detections.

https://play.google.com/store/apps/details?id=com.radiusnetworks.locate

hippiefahrzeug, 2015-04-11:
My apologies for posting the parser expression!

I'm sorry I didn't mention that: I also tried the app you mentioned (the old version locate v1.23 app and the new locate beacon 2.2), and also the reference app.

What happens:
- Moto X: in 'locate beacon' I see the beacon, and then it disappears. (it goes from 'immediate' to 'not detected', displaying 'not detected' 9 times out of 10 (or so)).

- Moo X: the old library (1.23) reliably shows the beacon once detected.

On HTC ONE it always works.

Btw: I also tried setAndroidLScanningDisabled(true) in my own code. That had no effect. I will add some more devices to the test.

I will attach a log file as soon as I get back to it. Thanks for your quick answer.

hippiefahrzeug, 2015-04-13:
Here is a log file i just created: https://gist.github.com/hippiefahrzeug/1bad13fda30aa9191bb9

davidgyoung, 2015-04-13:
This log confirms that the bluetooth APIs are rarely picking up those beacons.  

The fact that you report it works on the old library reliably is most curious, as the bluetooth detection part is hardly changed.

Can you please confirm where you set `setAndroidLScanningDisabled(true)` in your code?  This needs to be put immediately after the beaconManager is first created to be effective.  I mention this again because the Android 5 scanning APIs are a prime candidate for explaining this issue.  Unfortunately I have no Moto X to test with.

hippiefahrzeug, 2015-04-13:
actually, I took out setAndroidLScanningDisabled(true) in my previous attempt so that I'm as close to defaults as possible. here's a new log with this call, right at the very beginning: https://gist.github.com/hippiefahrzeug/01d112c52a7396ef55b5

davidgyoung, 2015-04-13:
Thanks, @hippiefahrzeug that is very helpful.  From this log, I can see that there were 47 scan cycles (1.1 seconds by default, which would be a total log file period of 51.7 seconds.)

During this time there were only 10 times where the beacon was detected by the Android Bluetooth APIs, as shown by lines like this:

D/BluetoothLeScanner(29322): onScanResult()

It looks like there is a 10 second period in the middle of the log where there were no detections at all (no lines like above.)  

Would it be possible for you to run a test with debug logging on the same phone hardware, with the same beacon transmitting, but with the old library you mentioned working fine?

hippiefahrzeug, 2015-04-13:
I see now what happens: I think it behaves the same with the old library! I just checked the log of the old locate beacon app and this is what I see: 

 % ~/adt-bundle-mac-x86_64-20130522/sdk/platform-tools/adb logcat | grep "I see"
I/RangingActivity( 7986): I see: 0 iBeacons
I/RangingActivity( 7986): I see: 0 iBeacons
I/RangingActivity( 7986): I see: 0 iBeacons
I/RangingActivity( 7986): I see: 1 iBeacons
I/RangingActivity( 7986): I see: 0 iBeacons
I/RangingActivity( 7986): I see: 0 iBeacons
I/RangingActivity( 7986): I see: 0 iBeacons
I/RangingActivity( 7986): I see: 0 iBeacons
I/RangingActivity( 7986): I see: 1 iBeacons
I/RangingActivity( 7986): I see: 0 iBeacons
I/RangingActivity( 7986): I see: 0 iBeacons
I/RangingActivity( 7986): I see: 0 iBeacons
I/RangingActivity( 7986): I see: 0 iBeacons

So... more or less the same pattern

What confused me: In the app UI, it doesn't clear the result as it does in the new app, which made me believe that the app still sees the beacon.

I also went back with my code to try it out myself with the old code base, and you wouldn't believe it: it is still unreliable! I think there is nothing wrong with the library after all.

hippiefahrzeug, 2015-04-13:
Ah I see you wrote at the same time as I was trying it out with the old library. Indeed I can confirm that I was hunting for a red herring. Sorry about that, I think we can close this issue.

davidgyoung, 2015-04-13:
Well, that's bad news in terms of being able to fix it in this library, as it points to a problem with scanning on the Moto X.   Just to eliminate any code that could be causing this inside the Android Beacon Library, can you please try running the code below, and capturing the log output on both the Moto X and the HTC One?  If we can demonstrate that this simple code does not work properly on the Moto X, perhaps we can get somebody at Motorola to comment on this.

Also, it would be very useful to know the OS build number from Settings -> About Phone -> Build number.

```
    private BluetoothAdapter mBluetoothAdapter;
    private static final String TAG = "MainActivity";

    @Override
    protected void onCreate(Bundle savedInstanceState) {
        super.onCreate(savedInstanceState);
        setContentView(R.layout.activity_main);
        mBluetoothAdapter = ((BluetoothManager) getSystemService(Context.BLUETOOTH_SERVICE)).getAdapter();
        mBluetoothAdapter.startLeScan(new BluetoothAdapter.LeScanCallback() {
            @Override
            public void onLeScan(final BluetoothDevice device, int rssi, byte[] scanRecord) {
                Log.d(TAG, "Scanned BLE device with mac: "+device.getAddress() );
            }
        });

    }
```

hippiefahrzeug, 2015-04-13:
ah, that is an interesting way of looking at it. I didn't try it out at that level, but now I extended your code up there a little bit so I stop the scan and start it again. Similarly as you do it in CycledLeScannerForJellyBeanMr2, but not with a timeout but I wait for the callback to happen. Like so:

        mBluetoothAdapter.startLeScan(new BluetoothAdapter.LeScanCallback() {
            @Override
            public void onLeScan(final BluetoothDevice device, int rssi, byte[] scanRecord) {
                Log.d(TAG, "Scanned BLE device with mac: " + device.getAddress());
                long e = System.currentTimeMillis();
                Log.d(TAG, "in " + (e-s) + "ms");
                mBluetoothAdapter.stopLeScan(this);
                s = System.currentTimeMillis();
                mBluetoothAdapter.startLeScan(this);
            }
        });

and this is what I see on HTC ONE:

    D/############( 9716): Scanned BLE device with mac: 20:CD:39:A4:A3:AE
    D/############( 9716): in 559ms
    D/############( 9716): Scanned BLE device with mac: 20:CD:39:A4:A3:AE
    D/############( 9716): in 128ms
    D/############( 9716): Scanned BLE device with mac: 20:CD:39:A4:A3:AE
    D/############( 9716): in 315ms
    D/############( 9716): Scanned BLE device with mac: 20:CD:39:A4:A3:AE
    D/############( 9716): in 136ms
    D/############( 9716): Scanned BLE device with mac: 20:CD:39:A4:A3:AE
    D/############( 9716): in 379ms
    D/############( 9716): Scanned BLE device with mac: 20:CD:39:A4:A3:AE
    D/############( 9716): in 355ms
    D/############( 9716): Scanned BLE device with mac: 20:CD:39:A4:A3:AE
    D/############( 9716): in 161ms
    D/############( 9716): Scanned BLE device with mac: 20:CD:39:A4:A3:AE
    D/############( 9716): in 328ms
    D/############( 9716): Scanned BLE device with mac: 20:CD:39:A4:A3:AE
    D/############( 9716): in 211ms
    D/############( 9716): Scanned BLE device with mac: 20:CD:39:A4:A3:AE
    D/############( 9716): in 152ms

... and this on the Moto X:

    D/############(25608): Scanned BLE device with mac: 20:CD:39:A4:A3:AE
    D/############(25608): in 4566ms
    D/############(25608): Scanned BLE device with mac: 20:CD:39:A4:A3:AE
    D/############(25608): in 240ms
    D/############(25608): Scanned BLE device with mac: 20:CD:39:A4:A3:AE
    D/############(25608): in 5088ms
    D/############(25608): Scanned BLE device with mac: 20:CD:39:A4:A3:AE
    D/############(25608): in 3290ms
    D/############(25608): Scanned BLE device with mac: 20:CD:39:A4:A3:AE
    D/############(25608): in 1540ms
    D/############(25608): Scanned BLE device with mac: 20:CD:39:A4:A3:AE
    D/############(25608): in 2524ms
    D/############(25608): Scanned BLE device with mac: 20:CD:39:A4:A3:AE
    D/############(25608): in 2559ms
    D/############(25608): Scanned BLE device with mac: 20:CD:39:A4:A3:AE
    D/############(25608): in 5096ms
    D/############(25608): Scanned BLE device with mac: 20:CD:39:A4:A3:AE
    D/############(25608): in 10190ms
    D/############(25608): Scanned BLE device with mac: 20:CD:39:A4:A3:AE
    D/############(25608): in 13983ms
    D/############(25608): Scanned BLE device with mac: 20:CD:39:A4:A3:AE
    D/############(25608): in 240ms
    D/############(25608): Scanned BLE device with mac: 20:CD:39:A4:A3:AE
    D/############(25608): in 4830ms
    D/############(25608): Scanned BLE device with mac: 20:CD:39:A4:A3:AE
    D/############(25608): in 512ms
    D/############(25608): Scanned BLE device with mac: 20:CD:39:A4:A3:AE
    D/############(25608): in 4712ms
    D/############(25608): Scanned BLE device with mac: 20:CD:39:A4:A3:AE
    D/############(25608): in 244ms
    D/############(25608): Scanned BLE device with mac: 20:CD:39:A4:A3:AE
    D/############(25608): in 500ms
    D/############(25608): Scanned BLE device with mac: 20:CD:39:A4:A3:AE
    D/############(25608): in 4319ms
    D/############(25608): Scanned BLE device with mac: 20:CD:39:A4:A3:AE
    D/############(25608): in 508ms
    D/############(25608): Scanned BLE device with mac: 20:CD:39:A4:A3:AE
    D/############(25608): in 4592ms
    D/############(25608): Scanned BLE device with mac: 20:CD:39:A4:A3:AE
    D/############(25608): in 508ms
    D/############(25608): Scanned BLE device with mac: 20:CD:39:A4:A3:AE
    D/############(25608): in 9689ms
    D/############(25608): Scanned BLE device with mac: 20:CD:39:A4:A3:AE
    D/############(25608): in 5065ms

This is quite interesting. The 1100ms timeout doesn't seem to be enough for the Moto X which exhibits rather slow scan cycles.

davidgyoung, 2015-04-13:
Well, it could be that it is just taking the Moto X a long time to find the beacon on average because it misses 90% of packets.  Just conjecture.

davidgyoung, 2015-04-14:
@hippiefahrzeug, can you please note the build number on the Moto X from Settings -> About Phone -> Build number?

hippiefahrzeug, 2015-04-14:
Build number on the Moto is LXE22.46-11

davidgyoung, 2015-04-14:
@hippiefahrzeug, two additional requests:

1. Can you comment on whether any other bluetooth processing is going on simultaneously on this device, and whether or not WiFi is active?  Obviously these are not solutions, but if you can reproduce the results with WiFi off and no other bluetooth processing going on it would be a cleaner test.

2. Would you be willing to capture a synchronized logcat excerpt alongside a snoop logs (Developer Options -> Enable HCI Snoop logging" -> adb pull /sdcard/btsnoop_hci.log) 

The latter could be useful for helping Motorola engineers investigate.

hippiefahrzeug, 2015-04-15:
Currently I'm on WiFi only (indeed wifi can have a big impact, e.g. on my nexus 4 or even HTC one with Android 4.x connectivity is terrible when bluetooth scanning was going on). I will do the tests as soon as I have cellular data.

davidgyoung, 2015-05-06:
Closing because this appears to be a Moto X firmware or chipset issue and not an issue with the library.  Based on the available info, it appears the Moto X cannot detect beacons properly if WiFi is active.

closed 2015-05-06 by davidgyoung

andersborgabiro, 2016-03-30:
I'm aware this is an old post, but do you have (or know of) a short-list of known devices with BLE and specifically beacon/broadcast problems?

davidgyoung, 2016-03-31:
Confirmed reports:  Nexus 4, Nexus 7, Moto G, Moto X.
Unconfirmed reports:  Samsung Galaxy S6 on Android 6.0.1 (no problem on 5.x):  http://stackoverflow.com/questions/36313143/android-6-marshmallow-wifi-bluetooth-sometimes-results-in-sockettimeoutexcep/36314952#36314952

andersborgabiro, 2016-03-31:
Excellent! Thanks.

Currently we have issues with Moto X, so hereby confirmed.

andersborgabiro, 2016-04-02:
Is there suspicion that the beacon library might be at fault in any of these cases?

Is Radius Locate using this library as-is, or do you have a more advanced library in-house?

davidgyoung, 2016-04-02:
A good engineer is *always* suspicious.   

That said, the main reason to think the problem is at a very low level is that when WiFi and BLE are in use simultaneously on a affected devices, low-level callbacks at the Android OS level for BLE advertisement detections slow down or stop.   On unaffected devices in use at the same time, this does not happen.   On affected devices, LogCat shows no operating system-level log lines for discovered BLE devices.  (Log lines which are annoying and impossible to turn off when everything is working correctly.)

Yes, the Locate App uses this exact library and doesn't add much more than user interface screens on top of it.

andersborgabiro, 2016-04-02:
Suspicious about ones own abilities maybe :).

The last paragraph is re-assuring.

I've noticed that BLE in Xperia Z3 sometimes can stop working, but it's enough to reactivate Bluetooth.