AltBeacon/android-beacon-library #164: Wifi not working after installing the app

Tushaarv, 2015-04-12:
I am encountering a strange issue, when app with this library is installed on some phones, WIFI stops working. Any suggestions?

davidgyoung, 2015-04-12:
Unfortunately, it is a known issue that on some devices WiFi and Bluetooth do not work simultaneously.  This is not a beacon-specific issue -- the same issue applies to bluetooth speakers and the like.  On such devices, simultaneous operation can cause the WiFi connection to fail, Bluetooth operation to fail, or both.  Examples of devices affected by this problem include the Nexus 4, Nexus 7 2013 edition and the Moto G.

You may wish to detect these device types and prompt the user to turn off WiFi for the app to operate properly.

See here:

https://github.com/AltBeacon/android-beacon-library/issues/49
https://github.com/Estimote/Android-SDK/issues/46

closed 2015-04-12 by davidgyoung

Tushaarv, 2015-04-15:
Thanks for the reply.