AltBeacon/android-beacon-library #165: App Crashes using BeaconSimulator crashes app when Android L APIs Active

davidgyoung, 2015-04-13:
App crashes with:  "Attempt to invoke virtual method 'android.bluetooth.le.BluetoothLeScanner android.bluetooth.BluetoothAdapter.getBluetoothLeScanner()' on a null object reference"

Reported by an end-user here:  http://stackoverflow.com/questions/29594996/issues-with-altbeacon-sample-running-on-simulator/29596488?noredirect=1#comment47340114_29596488

davidgyoung, 2015-04-16:
To fix this, the CycledLeScannerForLollipop needs to not crash if Bluetooth is not available.

ankitgupta100, 2015-04-20:
Hi David, for now I have implemented this function in my code - 

     public boolean IsBLESupportedOnDevice(Context context) {
         if (Build.VERSION.SDK_INT >= 19 && context.getPackageManager().hasSystemFeature(PackageManager.FEATURE_BLUETOOTH_LE))    
         {
              Log.d(TAG, "BLE is supported, so need to disable L scanning");
              return true;
         }
         return false;
     }

If it returns true, I am adding below line in the code -

    beaconManager.setAndroidLScanningDisabled(true);

Can you verify if the API level and rest of the condition looks good?

davidgyoung, 2018-11-07:
Fixed in #751

closed 2018-11-07 by davidgyoung