AltBeacon/android-beacon-library #168: IntentService/Service/CustomClass As BeaconConsumer

NabilNoaman, 2015-04-16:
Hi ,

I want to ask about samples for using BeaconConsumer with 
- IntentService or Service 
- CustomClass(neither Activity nor Service ).

It was shown in [BeaconManager] (https://altbeacon.github.io/android-beacon-library/javadoc/org/altbeacon/beacon/BeaconManager.html) documentation we can use it with **Service**  
"An class used to set up interaction with beacons from an Activity or Service"

I tried to implement IntentService As BeaconConsumer but I got diffrent error:

Service has leaked ServiceConnection org.altbeacon.beacon.BeaconManager$1@42c6f6b0 that was originally bound here

I found similar issue [here] (http://stackoverflow.com/questions/26323465/radiusnetwork-altbeacon-leaking-serviceconnection/29669038#29669038) but do not understand answer well.

davidgyoung, 2015-04-16:
I agree that posted samples of this kind of usage would be helpful and are not available at this time.  I will leave this issue open until these are added.   

In the meantime, if you would like help with your specific case, please post a question on strackoverflow.com including parts of your IntentService code

ccoulton, 2015-08-14:
These examples would be really great.... I've tried for serval days to get beacons working for our android SDK; and while they work when bind is pointed at Activity that implements a 'beaconconsumer'; and then the ranging and monitoring 'callbacks' can be in another class binding to a service or intent service or any other object.  I've even tried customizing the binding and starting the service manually but it still will not bind.

davidgyoung, 2021-08-06:
closed per #1046 with the deprecation of BeaconConsumer.  Samples now work outside an Activity

closed 2021-08-06 by davidgyoung