AltBeacon/android-beacon-library #172: crash on Android 5.0.1 Samsung device

quitequick, 2015-04-17:
Hi - I just got a crash from AltBeacon 2.1.4 on a Samsung device :

Build.VERSION.RELEASE 5.0.1
Build.VERSION.SDK_INT 21
Build.MANUFACTURER samsung
Build.MODEL SM-N910F
Build.PRODUCT trltexx
Build.DEVICE trlte

With stacktrace :

java.util.concurrent.RejectedExecutionException: Task android.os.AsyncTask$3@180d6bbf rejected from java.util.concurrent.ThreadPoolExecutor@2945428c[Running, pool size = 9, active threads = 9, queued tasks = 127, completed tasks = 2925]
	at java.util.concurrent.ThreadPoolExecutor$AbortPolicy.rejectedExecution(ThreadPoolExecutor.java:2011)
	at java.util.concurrent.ThreadPoolExecutor.reject(ThreadPoolExecutor.java:793)
	at java.util.concurrent.ThreadPoolExecutor.execute(ThreadPoolExecutor.java:1339)
	at android.os.AsyncTask.executeOnExecutor(AsyncTask.java:590)
	at org.altbeacon.beacon.service.BeaconService$1.onLeScan(BeaconService.java:300)
	at org.altbeacon.beacon.service.scanner.CycledLeScannerForLollipop$2.onScanResult(CycledLeScannerForLollipop.java:234)
	at android.bluetooth.le.BluetoothLeScanner$BleScanCallbackWrapper$1.run(BluetoothLeScanner.java:347)
	at android.os.Handler.handleCallback(Handler.java:739)
	at android.os.Handler.dispatchMessage(Handler.java:95)
	at android.os.Looper.loop(Looper.java:145)
	at android.app.ActivityThread.main(ActivityThread.java:5944)
	at java.lang.reflect.Method.invoke(Native Method)
	at java.lang.reflect.Method.invoke(Method.java:372)
	at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1399)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1194)

Any ideas?

Thanks!

davidgyoung, 2015-04-17:
This was fixed here:  https://github.com/AltBeacon/android-beacon-library/commit/901ef18482ddd746fc3f5fc3b4b3c99ba1908eec

It did not make it into the 2.1.4 release, but will go into the next one.  This bug was introduced in 2.1, so you can avoid it for now if you roll back to 2.0.

closed 2015-04-17 by davidgyoung

quitequick, 2015-04-20:
Hi - thanks for letting me know. Do you have an estimation on when the next release will be made (including the fix)?

Thanks!

eckoj433, 2015-06-26:
When im using the library in a native module for titanium, this bug seems to occur much more frequently, especially after closing the app and opening it when the service is already runnning in the background. Going back to 2.0.0 fixed the issue, but you should consider this fix pretty critical, and release a new version asap.

davidgyoung, 2015-06-26:
@eckoj433 this change is in this release:

https://github.com/AltBeacon/android-beacon-library/releases/tag/2.3

Which is the latest on JCenter and Maven.  Are you still seeing this problem in 2.3?

eckoj433, 2015-06-26:
@davidgyoung have not tested with 2.3, was checking http://altbeacon.github.io/android-beacon-library/download.html for new official releases, so didnt realise that 2.3 was out yet.

davidgyoung, 2015-06-26:
Oops!  Need to update that page.  Thanks for alerting me.

davidgyoung, 2015-06-26:
done