AltBeacon/android-beacon-library #173: Long data fields are not parsed properly

davidgyoung, 2015-04-24:
Data fields don't parse properly if they are:
 * little endian
 * are longer than four bytes

wns349, 2016-04-28:
Hello, 
I was wondering if there's any update on this. 

It seems that the `BeaconParser` is still not able to parse data fields correctly.

My custom layout is as follow: (notice data field of size 8)
```
BeaconParser parser = new BeaconParser();
parser.setBeaconLayout("m:2-3=0118,i:4-7,p:8-8,d:9-16,d:18-25");
```

It seems that `BeaconParser` is incorrectly parsing the data string (in hexadecimal format) to Long.

davidgyoung, 2016-04-28:
There has been no work on this no.  In fact, since I noticed the problem, you are the only person I have heard from who seems to  be interested in the issue.

The data field parsing needs to be re-implemented to populate an arbitrary byte array provide new accessor methods that can get the data fields in this format.

Until then, the data fields are essentially limited to four bytes, and accessors will always return the bytes as as a big-endian signed long integer.

I'd welcome a pull request if you are interested in working on this.  A change should leave the existing accessor methods as-is for backward compatibility, but provide new accessors that provide access to byte arrays.