AltBeacon/android-beacon-library #176: use of AlarmManager.ELAPSED_REALTIME_WAKEUP in CycledLeScanner

quitequick, 2015-05-06:
In CycledLeScanner.setWakeUpAlarm() I just noticed this :

        alarmManager.set(AlarmManager.ELAPSED_REALTIME_WAKEUP, System.currentTimeMillis() + milliseconds, mWakeUpOperation);

If I understand the AlarmManager docs correctly, AlarmManager.ELAPSED_REALTIME_WAKEUP requires a trigger time to be specified as a time relative to the last reboot time. So to trigger an alarm in 'milliseconds' from now using AlarmManager.ELAPSED_REALTIME_WAKEUP you would use :

        alarmManager.set(AlarmManager.ELAPSED_REALTIME_WAKEUP, SystemClock.elapsedRealtime() + milliseconds, mWakeUpOperation);

As it stands, it looks like the alarm would trigger some time in the very distant future.

lpellis, 2015-05-06:
I reported this a while back, the fix was made dont know when it will be part of release.
https://github.com/AltBeacon/android-beacon-library/issues/154

davidgyoung, 2015-05-06:
closing as duplicate

closed 2015-05-06 by davidgyoung