AltBeacon/android-beacon-library #179: 5 minutes is not configurable in setWakeUpAlarm

yankeppey, 2015-05-13:
Hello,
right now the time between wake up alarms can't be less than 5 minutes:

        // wake up time will be the maximum of 5 minutes, the scan period, the between scan period         long milliseconds = 1000l * 60 * 5; /* five minutes */

Is there a reason for this particular value? 
Can you probably make it configurable?

davidgyoung, 2015-05-14:
@yankeppey, can you please explain the use case for making this configurable?  The purpose of the wake up alarm is to restart scanning if the app stops running due to memory pressure or some other reason before the next schedules scan time starts.

If this were configurable, I'm not sure how we would explain the purpose of the configuration setting.

yankeppey, 2015-05-14:
I think the purpose of the wake up alarm is to ensure that device continues to scan beacons periodically even being in sleep mode.
Right now, as I got it, there are three cases:
- foreground mode (i.e. at least one activity is in resumed state) - device scans beacons with period defined by foregroundBetweenScanPeriod, using handler to post delayed scans
- background mode (all activity are in paused state) - device scans beacons with period defined by backgroundBetweenScanPeriod, using handler to post delayed scans
- sleep mode (user turned off the screen, no wake locks are hold) - everything stops, device wakes up and scans beacons every 5 minutes (that value is hardcoded).
And the last case is pretty common - the most of the time the phone sleeps in user's pocket. 
I want the phone to vibrate-notify if the user comes in area of a particular beacon, that provides, for example, some deal/discount, and 5 minutes is pretty long period for this (user can go though the whole beacon area without any notification). 
Please correct me if I'm wrong somewhere.

davidgyoung, 2015-05-16:
The wakeup alarm is really used as a backup.  The regular cycling of scans in the background and foreground uses a `postDelayed` method on a `Handler`, because this is a lighter weight alternative to using `AlarmManager` for frequently occurring events (e.g. scanning every 1.1 seconds.)  So unless the Android OS stops the application due to memory pressure or doing into deep sleep, scanning will happen at configurable intervals set by the four methods:

    beaconManager.setForegroundScanPeriod
    beaconManagersetForegroundBetweenScanPeriod
    beaconManagersetBackgroundScanPeriod
    beaconManagersetBackgroundBetweenScanPeriod

The only time the AlarmManager kicks in is after the app is killed due to memory pressure or due to doing into deep sleep.  When it fires, it will resume periodically scanning for beacons using the `Handler` `postDelayed` technique at the configured rates above.

This does not mean that it only scans for beacons every 5 minutes, it just means that there could be a maximum window of 5 minutes after the app is killed do to memory pressure or going into deep sleep before it resumes scanning at configurable rates.  The reason that this is not shorter is because if the user is doing something on the phone that is very memory intensive, it is probably not reasonable to immediately keep restarting background scanning when the device is short on memory.  The 5 minute minimum is intended to give time for the memory intensive operation to complete.

pollux-, 2015-05-20:
@davidgyoung  Assume device in deep sleep and AlarmManager kicks in wake up the CPU , application still in background,  App going to respect beaconManagersetBackgroundScanPeriod and
beaconManagersetBackgroundBetweenScanPeriod ??.  Any other way to create Bluetooth LE Service that will be running all the time even when the CPU sleeps and respecting the battery life ??  How the google fitness app collect data(step count), even when my phone is deep sleep ??

davidgyoung, 2015-05-26:
@pollux- good question.  I don't know the answer, but would be open to other techniques for collecting such data in the background when the phone is in deep sleep.

pollux-, 2015-06-04:
@davidgyoung  Thanks for your response..
If you got anything regarding that,  please keep me posted. I really appreciate that..