AltBeacon/android-beacon-library #184: Crash Samsung Galaxy S6 Android 5 set alarm

duffinator, 2015-05-18:
There is a problem with the Samsung Galaxy S6 with Android 5. If an app sets the alarm with the alarmmanager the app can crash. There is also a post in the Samsung developer forum: http://developer.samsung.com/forum/board/thread/view.do?boardName=General&messageId=280286&listLines=15&startId=zzzzz~&searchSubId=0000000001

Here ist a Crashlog:
Package: XXXXXXXXX
Version Code: 4703
Version Name: 4.703
Android: 5.0.2
Manufacturer: samsung
Model: SM-G920F
Date: Sat May 16 15:53:42 MESZ 2015

java.lang.SecurityException: !@Too many alarms (500) registered from pid 23706 uid 10235
	at android.os.Parcel.readException(Parcel.java:1540)
	at android.os.Parcel.readException(Parcel.java:1493)
	at android.app.IAlarmManager$Stub$Proxy.set(IAlarmManager.java:206)
	at android.app.AlarmManager.setImpl(AlarmManager.java:428)
	at android.app.AlarmManager.set(AlarmManager.java:215)
	at org.altbeacon.beacon.service.scanner.CycledLeScanner.setWakeUpAlarm(CycledLeScanner.java:304)
	at org.altbeacon.beacon.service.scanner.CycledLeScanner.setScanPeriods(CycledLeScanner.java:101)
	at org.altbeacon.beacon.service.BeaconService.setScanPeriods(BeaconService.java:293)
	at org.altbeacon.beacon.service.BeaconService$IncomingHandler.handleMessage(BeaconService.java:163)
	at android.os.Handler.dispatchMessage(Handler.java:102)
	at android.os.Looper.loop(Looper.java:145)
	at android.app.ActivityThread.main(ActivityThread.java:6141)
	at java.lang.reflect.Method.invoke(Native Method)
	at java.lang.reflect.Method.invoke(Method.java:372)
	at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1399)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1194)

davidgyoung, 2015-05-18:
Thanks for this report.  This bug will indeed affect library users, crashing apps on affected Samsung devices after four hours in the background..  Perhaps we can consider rescheduling Alarms instead of constructing new ones.

davidgyoung, 2015-05-19:
This is also reported here: 

http://stackoverflow.com/questions/29344971/java-lang-securityexception-too-many-alarms-500-registered-from-pid-10790-u/29692983#29692983

davidgyoung, 2015-05-19:
@duffinator, can you please try this beta release and see if it makes a difference?  I don't have an affected Samsung device to test on, but I have attempted to use the technique that reportedly works around the problem as reported in the above StackOverflow link.  

Please uninstall your app from your test device, then rebuild with the library version linked here and retest:

https://github.com/AltBeacon/android-beacon-library/releases/tag/2.3-beta3

Please let e know if this changes what you see.

duffinator, 2015-05-19:
The beta is not available in the maven repository. http://mvnrepository.com/artifact/org.altbeacon/android-beacon-library
Can you make it there available?

I see the changes but i had the problem already at another place in my project and there i have to catch the SecurityException. The FLAG_UPDATE_CURRENT didnt work for me.

duffinator, 2015-05-21:
I use the beta as aar file and i get the same crash. I think you have to catch the SecurityException

davidgyoung, 2015-05-21:
This is unfortunate.

Catching the security exception is OK as a last ditch effort to avoid a crash, but we really need to find a solution to make this continue working (e.g. prevent the security exception).

If the Alarms stop functioning on Samsung devices, then when the phone goes into deep sleep or the beacon app is killed by the OS due to memory pressure, there will be no way to restart scanning at a later time.  The Alarms are what solve that problem.

Obviously a crash is worse than the symptom I describe above.  I am just wary of fixing onevery serious problem by masking a slightly less serious problem.

duffinator, 2015-05-21:
I understand thats catching the security exception isnt the best way, but i think, that many alarms with the FLAG_CANCEL_CURRENT are registered it doesnt help to change this to FLAG_UPDATE_CURRENT. If the app is deinstalled and installed then this change will have an effect. I didnt tested it but thats my conclusion.

davidgyoung, 2015-05-21:
@duffinator, could you try de-installing and then re-installing your app compiled with the 2.3-beta3 library?  If you find this solves the problem, then it might give some other options.

duffinator, 2015-05-21:
Ok. I test it tomorrow. I found a way to get this crash. You have to write a for loop with 600 iterations and in this loop you have to set a alarm with the flag FLAG_CANCEL_CURRENT and catch the security exception. If you close the app, the library sets the alarm i get the crash. But thats only with a Samsung Galaxy S6 with Android 5.0.2 and bluetooth enabled.

duffinator, 2015-05-21:
Another thing: I use the 2.3-beta2 because there are the changes you made for this problem. In the 2.3-beta3 there is also FLAG_CANCEL_CURRENT and the method getWakeUpOperation() isnt there. Correct?
[image]

davidgyoung, 2015-05-21:
You want to test 2.3-beta3 after a clean install.  This version has the following change:

    -        alarmManager.set(AlarmManager.ELAPSED_REALTIME_WAKEUP, System.currentTimeMillis() + milliseconds, mWakeUpOperation);
    +        alarmManager.set(AlarmManager.RTC_WAKEUP, System.currentTimeMillis() + milliseconds, mWakeUpOperation);

davidgyoung, 2015-05-21:
@duffinator, I think you are correct that the 2.3-beta3 is accidentally missing the change you need.    I will roll a 2.3-beta4

davidgyoung, 2015-05-21:
The 2.3-beta4 release is available for download here:

https://github.com/AltBeacon/android-beacon-library/releases/tag/2.3-beta4

This has the proposed fix for the 500 Alarm limit on Samsung devices:

https://github.com/AltBeacon/android-beacon-library/pull/189

manpreetpatil, 2015-05-26:
@davidgyoung  I'm also getting same issue in my apps. Can you help me how can I fix it ?

davidgyoung, 2015-05-26:
@manpreetpatil, please try the 2.3-beta4 release linked above.  Try linking it with your app then running it to see if you still get the crash.  If you still get it, try uninstalling and reinstalling your app linked with 2.3-beta4 and test again.  (Re-installation may be needed to clear out the count of alarms started... this is still unclear.)

Please report your results along with your device make, model and OS version.

davidgyoung, 2015-05-29:
@duffinator and @manpreetpatil.  Any chance one of you can try the test mentioned in my above comment?  I'd like to get a new release out that fixes this problem, but I don't have a Samsung device with 5.0.2 to test the 2.3-beta4 release.

duffinator, 2015-06-02:
Sorry @davidgyoung I'am very busy, but this week i can test your release.

davidgyoung, 2015-06-18:
I have verified the change in 2.3-beta4 fixes this issue.

*  used a Samsung Galaxy S6 Edge with 5.0.2 from the [Samsung Remote Test Lab](http://developer.samsung.com/remotetestlab)
* I modified a copy of the reference application to restart alarms every 1.1 seconds while the app was ranging in the foreground.  I also modified it to track counts of the number of alarms restarted and the number of SecurityExceptions captured.  I modified the RangingActivity to display these two counts on the screen.

The end result showed that the count of alarm restarts climbed over 500 with no exceptions being thrown.

[image]
[image]

closed 2015-06-18 by davidgyoung